mbeddr/mbeddr.core #1164: Debugging inside OS callbacks

wuryel, 2015-02-19:
I would like to extend the mbeddr debugger to my extension of mbeddr. The "new" DSL shall provide access the the tasking facilities of the OS below. Right now I focus on a Linux host and use the pthread_create(...) directive to create and start a concurrent thread. This primitive expects a pointer
to a code snippet which is executed as a separate thread. Essentially the OS calls this code snippet whenever the thread is being scheduled.

Now I can place a breakpoint on DSL level into this code snippet and the debugger holds correctly but the stack frame construction fails when lifting the debugger information to DSL level.

The call stack information provided by the debugger goes below the DSL/Application level (so internals of the pthread_create(...) primitive are reported). This information (obviously) can't be mapped to the respective node on MPS tree level and the whole thing crashes.

I did some investigation in the mbeddr debugger implementation and identified a few checks that I removed in order to be able to successfully debug.

In the current implementation exceptions are thrown if it is not possible to map debugger information to MPS nodes. After my tiny changes these exceptions are not thrown anymore, but there is just no mapping performed. Maybe this is not the intended or best way to solve the issue (by far I haven't understood all the implementation), but at least it allows debugging code inside an OS callback routine.

This is what I did:

1) Source files refered to by the debug info are allowed to have the extension ".S" and ".s" besides just ".c". So this had to be added in these three operations

1a)  -- com.mbeddr.core.debug.(util models).debugger.(mapper).(state).(frame-mapper).(impl).MFrameMapperImpl.getFileName()
1b)  -- com.mbeddr.core.debug.(util models).debugger.(mapper).(state).(frame-mapper).(impl).GeneratedCFilePath.GeneratedCFilePath()
1c)  -- com.mbeddr.core.debug.(util models).debugger.(mapper).(state).(frame-mapper).(impl).GeneratedCFilePath.getUnitName

2) If the stack frame tracing doesn't work, this doesn't necessarily mean that we have an issue. Maybe a warning is sufficient here, so the user is aware of the fact.
To solve this I just removed the statements throwing the respective exceptions:

2a)  -- com.mbeddr.core.debug.(util models).debugger.(mapper).(state).(frame-mapper).(impl).MFrameMapperImpl.getClosestNodeViaTracesNew()
    - within the else block the statement "throw new MMissingTracesException();" is removed;
    - instead "return null;" is inserted

2b) -- com.mbeddr.core.debug.(util models).debugger.(mapper).(state).(frame-mapper).(impl).MFrameMapperImpl.constructFrameMappings()
    - within the outer else block the statement "throw new MMissingTracesException("Could not establish the trace for C routine: " + mappingStackFrame.getStackFrame().routineName());" is removed

DomenikP, 2015-02-25:
>I would like to extend the mbeddr debugger to my extension of mbeddr. The "new" DSL shall provide access the the tasking facilities of the OS below. Right now I focus on a Linux host and use the pthread_create(...) directive to create and start a concurrent thread. This primitive expects a pointer
to a code snippet which is executed as a separate thread. Essentially the OS calls this code snippet whenever the thread is being scheduled.

This means, you would also expect the possibility to switch between threads in the debugger UI? Right now, we only support debugging of single-threaded applications, but building the multithread support is not a serious issue.

> Now I can place a breakpoint on DSL level into this code snippet and the debugger holds correctly but the stack frame construction fails when lifting the debugger information to DSL level.

> The call stack information provided by the debugger goes below the DSL/Application level (so internals of the pthread_create(...) primitive are reported). This information (obviously) can't be mapped to the respective node on MPS tree level and the whole thing crashes.

> I did some investigation in the mbeddr debugger implementation and identified a few checks that I removed in order to be able to successfully debug.

> In the current implementation exceptions are thrown if it is not possible to map debugger information to MPS nodes. After my tiny changes these exceptions are not thrown anymore, but there is just [...]

I totally agree, that it makes sense to debug on the C level, if mapping fails. In addition, I think it makes sense to have some kind of flag inside the run configuration, which controls wether informations should be mapped (same applies for stepping) or not. If this flag is not set, we just show the pure C call stack with all of its C variables. In case of stepping, we would also just use the stepping commands from the underlying C debugger.

What do you think about this solution?

>This is what I did:

>1) Source files refered to by the debug info are allowed to have the extension ".S" and ".s" besides just ".c". So this had to be added in these three operations

>1a) -- com.mbeddr.core.debug.(util models).debugger.(mapper).(state).(frame-mapper).(impl).MFrameMapperImpl.getFileName()
1b) -- com.mbeddr.core.debug.(util models).debugger.(mapper).(state).(frame-mapper).(impl).GeneratedCFilePath.GeneratedCFilePath()
1c) -- com.mbeddr.core.debug.(util models).debugger.(mapper).(state).(frame-mapper).(impl).GeneratedCFilePath.getUnitName

>2) If the stack frame tracing doesn't work, this doesn't necessarily mean that we have an issue. Maybe a warning is sufficient here, so the user is aware of the fact.
To solve this I just removed the statements throwing the respective exceptions:

>2a) -- com.mbeddr.core.debug.(util models).debugger.(mapper).(state).(frame-mapper).(impl).MFrameMapperImpl.getClosestNodeViaTracesNew()
- within the else block the statement "throw new MMissingTracesException();" is removed;
- instead "return null;" is inserted

>2b) -- com.mbeddr.core.debug.(util models).debugger.(mapper).(state).(frame-mapper).(impl).MFrameMapperImpl.constructFrameMappings()
- within the outer else block the statement "throw new MMissingTracesException("Could not establish the trace for C routine: " + mappingStackFrame.getStackFrame().routineName());" is removed

Sounds good! :-) 

I would like to review your changes in detail and integrate them into mbeddr. Can you please send us a pull request.

wuryel, 2015-02-25:
Ok, I sent the pull request - hope I didn't mess it up ;-)

1) Having multi-threaded applications supported by the debugger would be great, of course. Although with many embedded platforms this is not supported by the hardware/gbd. My intention was to be able to debug at all, regardless of the thread.

2) I like your idea with the flag in the run configuration.

DomenikP, 2015-02-25:
I'm sorry, but since we're an Eclipse Project, you will need to sign an Eclipse Contributor License Agreement. I didn't consider this when writing the previous post. :-(

The following page explains the details:
https://wiki.eclipse.org/Development_Resources/Handling_Git_Contributions

DomenikP, 2015-02-25:
You can create your Eclipse account (and sign the Eclipse CLA) here:
https://dev.eclipse.org/portal/myfoundation/portal/portal.php

wuryel, 2015-02-25:
ok, I created and signed ;-)

DomenikP, 2015-03-05:
I just merged your change into mbeddr. 

The code is currently on a feature branch and will be built by our CI environment. In case everything stays green, I will merge it into master later today.

One comment on debugging in .S files:
I removed validations that check for file endings (e.g., .c or .S), as I don't think we really needed them.

DomenikP, 2015-03-10:
merged

closed 2015-03-10 by DomenikP