mbeddr/mbeddr.core #1170: Using MPS'es ANT for the build

qutorial, 2015-03-24:
There was a discussion to stop using the system OS-dependent ant in the build process, and switch to the ant shipping in MPS. It was decided to switch to it after a release, in the end of April.

> [Tuesday, 24 March 2015 15:53:46 Zaur Molotnikov] There is one interesting idea, which Alex suggested to me. This is an alternative way to fix the build, and might be a good idea on its own actually.  MPS comes with ant AFAU. So the idea would be to use this exactly ant, and not the one installed in the system. This is nice because: the user does not have to install ant, and because on all the platforms the build process will use the same ant. What do you think of it? 

This would prevent bugs like this:
https://groups.google.com/forum/#!topic/mbeddr-discuss/or3DELdBLeM
https://youtrack.jetbrains.com/issue/MPS-21734

Assigned to Domenik to get acquainted with it, as Kolja and me have discussed it and know about it already.

qutorial, 2015-03-24:
Currently the installer checks for the bug and recommends the solution
http://mbeddr.fortiss.org/download/

Installer modification will be needed, once this issues is solved.

alexanderpann, 2022-04-27:
Closing as not relevant anymore (we use gradle).

closed 2022-04-27 by alexanderpann