mbeddr/mbeddr.core #1171: Struct and Union Type Designated Initializers

szabta89, 2015-03-30:
The designated initializer feature in mbeddr should be reworked because various Exceptions are thrown when you play around with such a thing. A concrete example is this one:

struct Person {
	int age;
	int children;
	PersonAddress address;
};

struct PersonAddress {
	int door;
	char* street;
};

struct Person p3 = {
	.age = 3,
	.children = 10,
	.address.door = 10,
	.address.street = null
};

The initializer with the dot expression cannot be typed in.

coolya, 2015-03-30:
mbeddr does not use the dot as a prefix for named initializations of structs. Simply type the membername and it works.

coolya, 2015-03-30:
Do you have stack traces for the exceptions? I played a round with it and I was unable produce any.

szabta89, 2015-03-30:
Indeed, the initializers can be used this way, I didn't know that:

Person p3 = { 
  age = 25, 
  children = 0, 
  address = { 
    door = 10, 
    street = "" 
  } 
};

I don't have an exact trace right now. ArrayIndexOutOfBounds Exceptions are thrown when the getTypeCalculationNodeForElement methods (both in struct and uniontype) are called. I think this is related to the fact that the members() method is used, not the full members list of the type. The members() method call filters out the nested structs and unions.