mbeddr/mbeddr.core #1174: Build master fails with 'generate doesn't support the "createStaticRefs" attribute' on linux

maystar, 2015-04-19:
I've just tried to build the master version of mbeddr within linux. I tried it on Kubuntu 14.10, Kubuntu 15.04 RC (fresh virtual machine with just openjdk, git and ant installed) and openSUSE 13.2. I always got the following error: 
```
.../mbeddr.core/code/languages/build.xml:131: The following error occurred while executing this line:
.../mbeddr.core/code/languages/build.xml:213: The following error occurred while executing this line:
.../mbeddr.core/build/com.mbeddr.allScripts/build.xml:360: generate doesn't support the "createStaticRefs" attribute
```
I also try building with different jdks on one machine. It makes no difference if I use openjdk 7, openjdk 8 and oracle-jdk 7. Ant is installed with version 1.9.4.

markusvoelter, 2015-04-20:
Please use MPS 3.2.1. Then it should work

Markus 

Sent from a mobile device - please excuse brevity and typos.

maystar, 2015-04-20:
Hey Markus,
Thanks a lot for your quick response! I forgot to mention the MPS version. Indeed I've used 3.1.5 most of the time, because using the recent version lead to lots of errors. So I assumed you haven't migrated yet. But the source of the errors where just missing plugins. Now mbeddr builds fine so far. 
For me its quite inconvenient that the build script expects write permissions in the MPS program directory. I usually migrate plugins from one version to the next by copying the content of the user plugin directory. When MPS just changes in the minor version there is even no action necessary.

markusvoelter, 2015-04-20:
>
> So I assumed you haven't migrated yet. But the source of the errors where
> just missing plugins. Now mbeddr builds fine so far.
​ok.​

> For me its quite inconvenient that the build script expects write
> permissions in the MPS program directory.
​Does it? Why?​

Markus

> I usually migrate plugins from one version to the next by copying the
> content of the user plugin directory. When MPS just changes in the minor
> version there is even no action necessary.

bkolb, 2015-04-20:
2015-04-20 10:11 GMT+02:00 Markus Voelter <notifications@github.com>:

> > For me its quite inconvenient that the build script expects write
> > permissions in the MPS program directory.
> >
> ​Does it? Why?​

Yes, it does because the build script copies the spawner plugin into the
mps directory. they need to be installed unfortunately because it contains
a custom plugin.xml which is ignored otherwise.

Bernd

maystar, 2015-04-20:
And there are also lots of warnings like this:
java.io.FileNotFoundException: /tmp/MPS_3.2.1/system/log/mpsvcs.log
But anyway I close this bug.

closed 2015-04-20 by maystar