mbeddr/mbeddr.core #1178: MultilingualCommonUtil.findLocale problem when no country code is supplied

smmribeiro, 2015-05-12:
I found a problem in com.mbeddr.mpsutil.multilingual.common.runtime.plugin.MultilingualCommonUtil.
The following method has problems when countryCode is not supplied:

    public static Locale findLocale(string languageCode, string countryCode) {
      Locale.getAvailableLocales().asSequence.findFirst({~it => it.getLanguage() :eq: languageCode && (countryCode == null || it.getCountry() :eq: countryCode); }); 
    }

The following is what I get from the MPS Console:

    > MultilingualCommonUtil.findLocale("en", "GB")
      en_GB

    > MultilingualCommonUtil.findLocale("en", null) 
      en_MT 

    > Locale.getAvailableLocales().asSequence.where({~it => it.getLanguage() :eq: "en"; }) 
      [en_MT, en_GB, en_CA, en_US, en_ZA, en, en_SG, en_IE, en_IN, en_AU, en_NZ, en_PH] 

Basically, when looking for a language without country code, it wrongly returns the first that it encounters.
The test should check for the empty country code... something like:

    countryCode == null && it.getCountry().isEmpty

sjka, 2015-05-12:
fixed with f1bc0c75c80e64d730eb02906211fe10bcb918ab
tagged as i2s_v41

closed 2015-05-12 by sjka

smmribeiro, 2015-05-12:
I can now get a language with no country code.
Thank you.