mbeddr/mbeddr.core #1181: Bad rendering of empty table: <no ...> row should span all columns

vlsi, 2015-05-24:
See how "<no arguments>" row does not span full row, so the table looks inconsistent:
[image]

Table editor: 
[image]

Row editor:
[image]

coolya, 2015-05-24:
Hi @vlsi,

we have a extra repository for the tables. Would you mind creating the issues over there? Would make them better to track for us.

https://github.com/slisson/mps-tables

vlsi, 2015-05-24:
Oh, got the idea. Will move there.

vlsi, 2015-05-24:
Moved to https://github.com/slisson/mps-tables/issues/91

closed 2015-05-24 by vlsi