mbeddr/mbeddr.core #1201: No way to specify values for instance parameters of composite components

slisson, 2015-07-10:
Here I declare the parameters

[image]

but when I instantiate the component, there is no way to define values for the parameters

[image]

bkolb, 2015-07-10:
this is not the way it is supposed to work.

you have to declare the parameters as fields in the component and then pass them into the instance config (e.g. in the setup runnable).

But I agree that this could be improved

markusvoelter, 2015-07-10:
Why is it possible to specifiy parameters for "internal instances" in the
first place?
​

closed 2015-07-10 by slisson

bkolb, 2015-07-10:
Because you may want to do some calculations in the setup runnable and then pass those into the initialize instances