mbeddr/mbeddr.core #1205: StackOverflowError in GeneratorUtil.ultimateEndpoint

slisson, 2015-07-15:
```
java.lang.StackOverflowError
	at java.util.concurrent.ConcurrentHashMap$Segment.put(ConcurrentHashMap.java:443)
	at java.util.concurrent.ConcurrentHashMap.putIfAbsent(ConcurrentHashMap.java:897)
	at jetbrains.mps.smodel.adapter.structure.MetaAdapterFactory.getProperty(MetaAdapterFactory.java:117)
	at jetbrains.mps.smodel.adapter.structure.MetaAdapterFactory.getProperty(MetaAdapterFactory.java:123)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.createNamespace(GeneratorUtil.java:208)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:181)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:182)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:182)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:182)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:182)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:182)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:182)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:182)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:182)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:182)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:182)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:182)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:182)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:182)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:182)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:182)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:182)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:182)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:182)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:182)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:182)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:182)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:182)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:182)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:182)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:182)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:182)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:182)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:182)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:182)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:182)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:182)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:182)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:182)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:182)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:182)
	at com.mbeddr.ext.compositecomponents.generator.flatteningCompComps.util.GeneratorUtil.ultimateEndpoint(GeneratorUtil.java:182)

```

bkolb, 2015-07-30:
Do you remember how the model looked like? From the code I don't understand why that happens

slisson, 2015-07-31:
There was a broken reference somewhere. It might be an MPS bug, that you can call behavior methods on "null". If you assume "instance" is null, then you will understand.