mbeddr/mbeddr.core #1208: Debugger: Cannot step into PortAdapterOpCallExpr

slisson, 2015-07-16:
I cannot step into the last "writeByte" call, but into the other two calls:
```
exported testcase testEncoder { 
  initialize ManchesterInstances; 
   
  encoderStream.writeByte(bin«00000000u»); 
  assert-equals(0) adapterBuffer.get() == bin«01010101u»; 
  assert-equals(1) adapterBuffer.get() == bin«01010101u»; 
   
  encoderStream.writeByte(bin«11111111u»); 
  assert-equals(2) adapterBuffer.get() == bin«10101010u»; 
  assert-equals(3) adapterBuffer.get() == bin«10101010u»; 
   
  encoderStream.writeByte(bin«11000101u»); 
  assert-equals(4) adapterBuffer.get() == bin«10100101u»; 
  assert-equals(5) adapterBuffer.get() == bin«01100110u»; 
   
} testEncoder(test case)
```

```
project: com.mbeddr.demo.robot (/Users/slisson/mps/mbeddr.core/code/applications/robot)
module:  com.mbeddr.demo.robot
model:   com.mbeddr.demo.robot.communication_tests
root:    MachesterTests [ImplementationModule]
node:    writeByte [PortAdapterOpCallExpr]
url:     http://localhost:8080/select/com.mbeddr.demo.robot/r:21fea1b4-4580-4b5b-b52d-a0b0fc2dc842/1086588841049765343/
```

slisson, 2015-07-16:
This only happens for parameter values from 194 to 253. When I use any value not in this range, I can step into the call.

alexanderpann, 2024-11-25:
The debugger was removed in https://github.com/mbeddr/mbeddr.core/pull/2509. The ticket is now obsolete.

closed 2024-11-25 by alexanderpann