mbeddr/mbeddr.core #1219: Clang / LLDB Support

sirinath, 2015-08-12:
Is it possible to support the Clang LLDB stack? Going forward LLVM will have better support in all platforms than GCC toolchain as companies like MS are also working on LLVM/Clang based tools:
https://github.com/dotnet/llilc
https://github.com/Microsoft/llvm
https://github.com/Microsoft/WinObjC
https://github.com/Microsoft/MIEngine
https://github.com/Microsoft/clang-tools-extra
https://github.com/Microsoft/clang

bkolb, 2015-08-13:
Yes, we would very much like to see that. Currently however, there is no support for LLDB in Eclipse and we reuse that part to integrate with GDB. As soon as support for LLDB is provided there we will also integrate it. As always: Contributions are very welcome!

coolya, 2016-11-14:
duplicate of #1147

closed 2016-11-14 by coolya