mbeddr/mbeddr.core #1227: "Validating tools are installed" dialog freezes when I try to run a main function from mbeddr

danielratiu, 2015-11-01:
sometimes I just cannot start a main function from mbeddr (see screenshot attached). I have paster also the stacktrace - it does not seem that we have deadlock.

[image]

2015-11-01 16:29:56
Full thread dump Java HotSpot(TM) 64-Bit Server VM (25.60-b23 mixed mode):

"RMI TCP Connection(2)-192.168.91.1" #3193 daemon prio=5 os_prio=0 tid=0x0000000021c4b000 nid=0x2828 runnable [0x000000004557e000]
   java.lang.Thread.State: RUNNABLE
	at java.net.SocketInputStream.socketRead0(Native Method)
	at java.net.SocketInputStream.socketRead(SocketInputStream.java:116)
	at java.net.SocketInputStream.read(SocketInputStream.java:170)
	at java.net.SocketInputStream.read(SocketInputStream.java:141)
	at java.io.BufferedInputStream.fill(BufferedInputStream.java:246)
	at java.io.BufferedInputStream.read(BufferedInputStream.java:265)
	- locked <0x0000000779ce6798> (a java.io.BufferedInputStream)
	at java.io.FilterInputStream.read(FilterInputStream.java:83)
	at sun.rmi.transport.tcp.TCPTransport.handleMessages(TCPTransport.java:550)
	at sun.rmi.transport.tcp.TCPTransport$ConnectionHandler.run0(TCPTransport.java:826)
	at sun.rmi.transport.tcp.TCPTransport$ConnectionHandler.lambda$run$256(TCPTransport.java:683)
	at sun.rmi.transport.tcp.TCPTransport$ConnectionHandler$$Lambda$32/365836524.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at sun.rmi.transport.tcp.TCPTransport$ConnectionHandler.run(TCPTransport.java:682)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)

   Locked ownable synchronizers:
	- <0x0000000778d26260> (a java.util.concurrent.ThreadPoolExecutor$Worker)

"JMX server connection timeout 3192" #3192 daemon prio=5 os_prio=0 tid=0x000000002769c000 nid=0x15d0 in Object.wait() [0x000000004543f000]
   java.lang.Thread.State: TIMED_WAITING (on object monitor)
	at java.lang.Object.wait(Native Method)
	at com.sun.jmx.remote.internal.ServerCommunicatorAdmin$Timeout.run(ServerCommunicatorAdmin.java:168)
	- locked <0x0000000778e1e998> (a [I)
	at java.lang.Thread.run(Thread.java:745)

   Locked ownable synchronizers:
	- None

"RMI Scheduler(0)" #3191 daemon prio=5 os_prio=0 tid=0x000000002eb5c800 nid=0xc9c waiting on condition [0x000000004533e000]
   java.lang.Thread.State: TIMED_WAITING (parking)
	at sun.misc.Unsafe.park(Native Method)
	- parking to wait for  <0x00000006cd78e328> (a java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject)
	at java.util.concurrent.locks.LockSupport.parkNanos(LockSupport.java:215)
	at java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject.awaitNanos(AbstractQueuedSynchronizer.java:2078)
	at java.util.concurrent.ScheduledThreadPoolExecutor$DelayedWorkQueue.take(ScheduledThreadPoolExecutor.java:1093)
	at java.util.concurrent.ScheduledThreadPoolExecutor$DelayedWorkQueue.take(ScheduledThreadPoolExecutor.java:809)
	at java.util.concurrent.ThreadPoolExecutor.getTask(ThreadPoolExecutor.java:1067)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1127)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)

   Locked ownable synchronizers:
	- None

"RMI TCP Connection(idle)" #3190 daemon prio=5 os_prio=0 tid=0x0000000030e8b000 nid=0x2a7c waiting on condition [0x0000000044cfe000]
   java.lang.Thread.State: TIMED_WAITING (parking)
	at sun.misc.Unsafe.park(Native Method)
	- parking to wait for  <0x00000006cd78e678> (a java.util.concurrent.SynchronousQueue$TransferStack)
	at java.util.concurrent.locks.LockSupport.parkNanos(LockSupport.java:215)
	at java.util.concurrent.SynchronousQueue$TransferStack.awaitFulfill(SynchronousQueue.java:460)
	at java.util.concurrent.SynchronousQueue$TransferStack.transfer(SynchronousQueue.java:362)
	at java.util.concurrent.SynchronousQueue.poll(SynchronousQueue.java:941)
	at java.util.concurrent.ThreadPoolExecutor.getTask(ThreadPoolExecutor.java:1066)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1127)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)

   Locked ownable synchronizers:
[... 2858 lines cut ...]

   Locked ownable synchronizers:
	- None

"VM Thread" os_prio=2 tid=0x000000000d480800 nid=0x200c runnable 

"GC task thread#0 (ParallelGC)" os_prio=0 tid=0x000000000201f000 nid=0x2154 runnable 

"GC task thread#1 (ParallelGC)" os_prio=0 tid=0x0000000002021000 nid=0x2768 runnable 

"GC task thread#2 (ParallelGC)" os_prio=0 tid=0x0000000002022800 nid=0x2204 runnable 

"GC task thread#3 (ParallelGC)" os_prio=0 tid=0x0000000002024000 nid=0x2520 runnable 

"GC task thread#4 (ParallelGC)" os_prio=0 tid=0x0000000002027800 nid=0x27d4 runnable 

"GC task thread#5 (ParallelGC)" os_prio=0 tid=0x000000000202a800 nid=0x20a8 runnable 

"GC task thread#6 (ParallelGC)" os_prio=0 tid=0x000000000202b800 nid=0x203c runnable 

"GC task thread#7 (ParallelGC)" os_prio=0 tid=0x000000000202d000 nid=0x239c runnable 

"VM Periodic Task Thread" os_prio=2 tid=0x000000000e5fd800 nid=0x2558 waiting on condition 

JNI global references: 1861

DomenikP, 2015-11-02:
Can you please be a bit more specific?

You right click on a main Function and select "Run ..". Next, the progress bars appear and MPS freezes?

danielratiu, 2015-11-02:
yes - MPS tries to validate the tools and that validation dialog does not
dissapear :-)

alexanderpann, 2024-07-14:
In case this is still an issue: The usage of `EventQueue#invokeAndWait` + `ProgressManager#runProcessWithProgressSynchronously` [here](http://127.0.0.1:63320/node?ref=r%3A35144171-bbda-419f-9015-4d1f075e1db4%28com.mbeddr.core.runconfiguration.pluginSolution.plugin%29%2F1288687059923023627&project=com.mbeddr.mpsutil) looks very dubious to me.