mbeddr/mbeddr.core #1228: Unintuitive generation from foreach statements

danielratiu, 2015-11-02:
the following statement is intended to initialize an array

foreach (clauses sized 10 as it) { it.clauseActive = true; }

... however, in the generated C code, we have:

for ( uint8_t __c = 0 ; __c < 10; __c++ )
  {
    clause_data_structure_clause_t_t __it = clause_data_structure_clauses[__c];
    __it.clauseActive = true;
  }

... thereby the assignment "it.clauseActive = true" is assigned to a temporary variable. ALL occurences of "it" on a left-hand-side of an assignment are thereby DOOMED

markusvoelter, 2015-11-03:
Which foreach is this? The one from the tutorial?

danielratiu, 2015-11-03:
no, it is the one from mbeddr

core.util.ForEachStatement

markusvoelter, 2015-11-03:
So I will change this to:

for ( uint8_t __c = 0 ; __c < 10; __c++ ) {
clause_data_structure_clause_t_t* __it = &clause_data_structure_clauses[__c];
(*__it).clauseActive = true;
}

danielratiu, 2015-11-03:
ok, this should do it

markusvoelter, 2015-11-03:
Fixed.

closed 2015-11-03 by markusvoelter