mbeddr/mbeddr.core #1232: MPS fails to load mbeddr library

wuryel, 2015-11-18:
After successfully compiling mbeddr, MPS refuses to load the mbeddr library.
Find attached the console log: [error.txt](https://github.com/mbeddr/mbeddr.core/files/37829/error.txt)

I'm working with
* MPS-3.3-EAP4-142.230
* JDK 1.7.0_75 (64bit)
* mbeddr tagged nightly-37
* linux (ubuntu) environment, all in user account

wuryel, 2015-11-18:
also raised an issue with jetbrains
https://youtrack.jetbrains.com/issue/MPS-23006

coolya, 2015-11-18:
The problem was introduce by @enikao in revision 3540bca205e645d6afbef2caab19f2ca7019a1a5 fixing it right now.

coolya, 2015-11-18:
Fixed in current master

wuryel, 2015-11-18:
sorry, this doesn't solve it.
The empty files are now removed and as such don't generate a failure anymore.
Nevertheless, the mbeddr library still doesn't load (same exception stack as reported above).
Interestingly the persistence version is set to 8 (instead of to 9) in mbeddr in the reported locations. 
[persistenceVersion.txt](https://github.com/mbeddr/mbeddr.core/files/38059/persistenceVersion.txt)
It was Markus' suggestion, that this could be related.
I changed all those to version 9, but still the same failure.

coolya, 2015-11-18:
Is this from the command line or within MPS?

If it's with in MPS please split your project libraries in two things:

`mbedder.core` pointing to `code/languages`

and 

`sl-all` pointing to `code/plugins/sl-all` 

I think the problem is that MPS tries to load some broken models somewhere in `code/applications`

coolya, 2015-11-18:
I have updated the instructions on the download page to reflect this changes. Please report back if it worked for you.

wuryel, 2015-11-19:
great! mbeddr loads now!
I'm working from within MPS, setting the libraries before loading any project in the dialog fields.

It is important to first load `sl-all` and than load `mbeddr.core`.
The order should be provided in the instructions on the download page.

thx

coolya, 2015-11-27:
Updated the ordering.

closed 2015-11-27 by coolya