mbeddr/mbeddr.core #1234: GrammarCells: Exception if abstract method is called

bkolb, 2015-11-20:
I have a BinaryExpression which is abstract and contains an abstract method `public static virtual int getPriority();` Binary Expression is a subconcept of Expression. 
The editor of BinaryExpression looks as follows:
```
projection: [- wrap % left % substitute grammar.constant wrap % right % -]  
grammar: rule: <derive from projection>    %left%  grammar.constant  %right%
```
`rule` calls getPriority() 

A VariableDeclaration has an init Expression. The editor looks as follows `[> # alias # % type % { name } optional [- = % init % -] <]``

when I type the `=` I get an exception because of the "missing" behavior method getPriority in the BinaryExpression.

Caused by: jetbrains.mps.core.aspects.behaviour.api.BHMethodNotFoundException: The method 'J(int)::getPriority([])VS[PUB]' could not be found within the 'com.mbeddr.spec.expression.structure.UnaryExpression BHDescriptor'.
	at com.mbeddr.spec.expression.behavior.UnaryExpression__BehaviorDescriptor.invokeSpecial0(UnaryExpression__BehaviorDescriptor.java:64)
	at jetbrains.mps.core.aspects.behaviour.BaseBHDescriptor.invokeSpecial(BaseBHDescriptor.java:358)
	at jetbrains.mps.core.aspects.behaviour.BaseBHDescriptor.invokeVirtual(BaseBHDescriptor.java:324)
	at jetbrains.mps.core.aspects.behaviour.BaseBHDescriptor.invoke(BaseBHDescriptor.java:222)
	at jetbrains.mps.core.aspects.behaviour.SMethodImpl.invoke(SMethodImpl.java:146)
	at com.mbeddr.spec.expression.editor.GrammarActionsDescriptor$38.getPriority(GrammarActionsDescriptor.java:1617)
	at com.mbeddr.mpsutil.grammarcells.runtime.RulesCache$4.select(RulesCache.java:80)
	at com.mbeddr.mpsutil.grammarcells.runtime.RulesCache$4.select(RulesCache.java:1)
	at jetbrains.mps.internal.collections.runtime.ISelector.invoke(ISelector.java:12)
	at jetbrains.mps.internal.collections.runtime.SelectComparator.compare(SelectComparator.java:23)
	at java.util.TimSort.binarySort(TimSort.java:292)
	at java.util.TimSort.sort(TimSort.java:217)
	at java.util.Arrays.sort(Arrays.java:1438)
	at jetbrains.mps.internal.collections.runtime.impl.SortingSequence.inputSortedWithSelector(SortingSequence.java:55)
	at jetbrains.mps.internal.collections.runtime.impl.SortingSequence.iterator(SortingSequence.java:40)
	at jetbrains.mps.internal.collections.runtime.ListSequence.fromIterable(ListSequence.java:122)
	at jetbrains.mps.internal.collections.runtime.AbstractSequence.toListSequence(AbstractSequence.java:215)
	at com.mbeddr.mpsutil.grammarcells.runtime.RulesCache.getRules(RulesCache.java:82)
	at com.mbeddr.mpsutil.grammarcells.runtime.Parser.getRules(Parser.java:500)
	at com.mbeddr.mpsutil.grammarcells.runtime.Parser.getRule(Parser.java:476)
	at com.mbeddr.mpsutil.grammarcells.runtime.Parser.isEndOf(Parser.java:617)
	at com.mbeddr.cc.requirements.franca.editor.GrammarActionsDescriptor$4.redirect(GrammarActionsDescriptor.java:128)
	at com.mbeddr.cc.requirements.franca.editor.GrammarActionsDescriptor.doGetSideTransformActions(GrammarActionsDescriptor.java:141)
	at com.mbeddr.mpsutil.grammarcells.runtime.AbstractGrammarActionDescriptor.getActions(AbstractGrammarActionDescriptor.java:34)
	at com.mbeddr.cc.requirements.franca.actions.QueriesGenerated.sideTransform_ActionsFactory_BaseConcept_994107119628465925(QueriesGenerated.java:34)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at jetbrains.mps.util.QueryMethodGenerated.invoke(QueryMethodGenerated.java:180)
	... 69 more

bkolb, 2015-11-20:
If I make getPriority not abstract it works as expected

slisson, 2015-11-20:
MPS should not allow to create static abstract methods. This is an MPS bug.

bkolb, 2015-11-20:
hm, ok

closed 2015-11-20 by bkolb