mbeddr/mbeddr.core #1237: Failing ant build script

Jochen-81, 2015-11-27:
I am running MPS3.2 on an Mac.
After cloning the git and changing the build.properties I run the buildLanguages.sh as descripted in the documentation. Having not much experience with ant I hope to find some help here. The script output looks like this : 

clean-and-build:

build-allScripts:

declare-mps-tasks:

fetchDependencies:

generate:
     [echo] generating
 [generate] Java HotSpot(TM) 64-Bit Server VM warning: ignoring option MaxPermSize=92m; support was removed in 8.0
 [generate] Library /Applications/MPS3.2.app/Contents/languages/languageDesign/jetbrains.mps.baseLanguage.lightweightdsl.jar does not exist.
 [generate] Library /Applications/MPS3.2.app/Contents/languages/languageDesign/jetbrains.mps.lang.customAspect.jar does not exist.
 [generate] Library /Applications/MPS3.2.app/Contents/languages/languageDesign/jetbrains.mps.lang.smodel.query.jar does not exist.
 [generate] Library /Applications/MPS3.2.app/Contents/lib/mps-environment.jar does not exist.
 [generate] Library /Applications/MPS3.2.app/Contents/plugins/mps-build/languages/build/jetbrains.mps.build.workflow.preset.jar does not exist.
 [generate] [      0]   INFO - ool.environment.MpsEnvironment - Creating MPS environment 
 [generate] [     20]   WARN - tbrains.mps.smodel.ModelAccess - Adding command listener to DefaultModelAccess: a command is the same as a write action 
 [generate] [     22]   WARN - tbrains.mps.smodel.ModelAccess - Adding command listener to DefaultModelAccess: a command is the same as a write action 
 [generate] [    111]   INFO - mps.library.LibraryInitializer - Adding libraries from jetbrains.mps.tool.builder.util.SetLibraryContributor 
 [generate] [    111]   INFO - mps.library.LibraryInitializer - Adding libraries from jetbrains.mps.tool.builder.util.SetLibraryContributor 
 [generate] [    116]   INFO - mps.library.LibraryInitializer - Loading 62 libraries 
 [generate] Generating: strict mode is on, parallel generation is on (4 threads), in-place is on, warnings are on, static references to replace dynamic is on
 [generate] Loaded module com.mbeddr.allInOne [solution]
 [generate] Loaded module com.mbeddr.analyses.tests.dev.build [solution]
 [generate] Loaded module com.mbeddr.build [solution]
 [generate] Loaded module com.mbeddr.cc.tests.dev.build [solution]
 [generate] Loaded module com.mbeddr.core.tests.build [solution]
 [generate] Loaded module com.mbeddr.core.tests.performance.build [solution]
 [generate] Loaded module com.mbeddr.debugger.testing.build.dev.build [solution]
 [generate] Loaded module com.mbeddr.debugger.tests.dev.build [solution]
 [generate] Loaded module com.mbeddr.ext.tests.build [solution]
 [generate] Loaded module com.mbeddr.platform [solution]
 [generate] Loaded module com.mbeddr.rcp [solution]
 [generate] Loaded module com.mbeddr.tutorial.dev.build [solution]
 [generate] Loaded module de.slisson.mps.all.build [solution]
 [generate] Loaded module jetbrains.mps.minimal [solution]
 [generate] Generating:
 [generate]     com.mbeddr.allInOne [solution]
 [generate]     com.mbeddr.analyses.tests.dev.build [solution]
 [generate]     com.mbeddr.build [solution]
 [generate]     com.mbeddr.cc.tests.dev.build [solution]
 [generate]     com.mbeddr.core.tests.build [solution]
 [generate]     com.mbeddr.core.tests.performance.build [solution]
 [generate]     com.mbeddr.debugger.testing.build.dev.build [solution]
 [generate]     com.mbeddr.debugger.tests.dev.build [solution]
 [generate]     com.mbeddr.ext.tests.build [solution]
 [generate]     com.mbeddr.platform [solution]
 [generate]     com.mbeddr.rcp [solution]
 [generate]     com.mbeddr.tutorial.dev.build [solution]
 [generate]     de.slisson.mps.all.build [solution]
 [generate]     jetbrains.mps.minimal [solution]
 [generate] Build not started: invalid make script
 [generate] Validation failed: No make facets requested, nothing to make
 [generate] [   2356]   INFO - .mps.make.service.CoreMakeTask - Other targets execution time: 0 ms;  
 [generate] Other targets execution time: 0 ms; 
 [generate] Build failed. See previous messages for details.
 [generate] java.lang.RuntimeException: ****************************************************************************************************
 [generate] 4 errors during generation:
 [generate] Build not started: invalid make script
 [generate] Validation failed: No make facets requested, nothing to make
 [generate] Build failed. See previous messages for details.
 [generate] Make was not successful
 [generate] ****************************************************************************************************
 [generate] 	at jetbrains.mps.tool.builder.MpsWorker.failBuild(MpsWorker.java:177)
 [generate] 	at jetbrains.mps.tool.builder.make.BaseGeneratorWorker.showStatistic(BaseGeneratorWorker.java:97)
 [generate] 	at jetbrains.mps.tool.builder.make.GeneratorWorker.work(GeneratorWorker.java:96)
 [generate] 	at jetbrains.mps.tool.builder.MpsWorker.workFromMain(MpsWorker.java:115)
 [generate] 	at jetbrains.mps.tool.builder.make.GeneratorWorker.main(GeneratorWorker.java:100)
 [generate] 	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
 [generate] 	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
 [generate] 	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
 [generate] 	at java.lang.reflect.Method.invoke(Method.java:497)
 [generate] 	at jetbrains.mps.tool.builder.AntBootstrap.main(AntBootstrap.java:30)
 [generate] 

BUILD FAILED
/Users/emrolab/Desktop/mbeddr/mbeddr.core/code/languages/build.xml:110: The following error occurred while executing this line:
/Users/emrolab/Desktop/mbeddr/mbeddr.core/code/languages/build.xml:188: The following error occurred while executing this line:
/Users/emrolab/Desktop/mbeddr/mbeddr.core/build/com.mbeddr.allScripts/build.xml:462: Process exited with code 1.

DomenikP, 2015-11-27:
Which branch/commit do you try to build?

Jochen-81, 2015-11-27:
I am trying to build the mbeddr.core.

DomenikP, 2015-11-27:
Ok, so you are most likely on mbeddr master.

Please install MPS 3.3 on your machine, update the MPS path in your build.properties and start again ant.

You can get the required MPS 3.3 installation from here:
https://confluence.jetbrains.com/display/MPS/JetBrains+MPS+EAP+Download+Page

Jochen-81, 2015-11-27:
Thank you for the advice, the build worked. I used 3.2 because it was labeled the last stable build.

closed 2015-11-27 by Jochen-81