mbeddr/mbeddr.core #1241: A few issues for the interpreter

markusvoelter, 2015-11-30:
different cases (guards)?
Was ist toggle Caching?
env.getRootEnvironment bleibt rot. Warum?
Macht es Sinn, bei den Type Mappings und der Type guard das $ schon einzufügen?

alexanderpann, 2025-04-07:
I'll try to answer some of those questions:

> different cases (guards)?
I guess this is implemented through typed child constraints.

> What is the toggle Caching?
`Enable Interpreter Cache` in the `Tools` main menu caches the lookup of the available interpreter execution classes. It does not refer to caching of interpreter results.

> env.getRootEnvironment stays red. Why?
I don't understand the question. This method is not used in the code anywhere anymore.

> type guard: I also want to check the node itself
I guess the workaround is to add an if condition to your interpreter and call `tryOtherInterpreter`. The question is if we want to enhance type child constraints or come up with a better solution.

> would be nice to be able to group them (as in the mapper)
Not sure if this refers to grouping of type guards (one evaluator can have more than one guard).

> Cannot change operator once one is entered (subsitute missing)
I fixed this issue.

enikao, 2025-04-08:
>> different cases (guards)?
> I guess this is implemented through typed child constraints.

I think no other guards than types work anymore. (`ConditionalInterpreter` is essentially useless)