mbeddr/mbeddr.core #1261: Optional leads to compile error

markusvoelter, 2016-01-04:
Replace the conditional for the constraint with an optional cell to see the exception:

project: org.iets3.core (/Users/markusvoelter/Documents/mbeddr/iets3.core/code/languages/org.iets3.core)
module:  org.iets3.components.functional
model:   org.iets3.components.functional.editor
node:    DataItemPortType_Editor [ConceptEditorDeclaration] (root)
url:     http://localhost:8080/select/org.iets3.core/r:b08adb63-6835-487f-a8d6-f6cefb8f131b/7804632404593474276/

[image]

slisson, 2016-01-05:
The problem is, that some editor components are inlined (cells from the editor component are copied into the generated model) and then the model contains concepts from languages that are not part of the generation plan.

A workaround is to add the languages ("querylist" and "tooltip" in your case) to "Languages engaged on generation".

slisson, 2016-01-07:
TODO: Show a warning or error

alexanderpann, 2023-02-16:
I am closing because this exact use case is working now.

closed 2023-02-16 by alexanderpann