mbeddr/mbeddr.core #1274: cc menu: Multiple entries / irrelevant entries

bkolb, 2016-01-19:
I have an if-statement. inside that i use the create target cc menu. I see entries e.g. for creating a local variable multiple times. It seems one is created nesting level (nested statement lists)

At the same place I also see a ForRangeStatement. I don't know why that shows up

slisson, 2016-01-20:
A ForRangeStatement is an INamedConcept and there is a ForRangeRef. Why is it a surprise, that it shown?

slisson, 2016-01-20:
Which one of the LocalVariableDeclarations do you what to see? Only the one for the closest ancestor StatementList?

bkolb, 2016-01-20:
> Why is it a surprise that the forRange is shown?
Becuase when you create it , it is out of scope. The whole forRange Statement is created before the reference. So it will look as follows
```
for (qweqqw ++ in [<left>..<right>[) { 
   
} for
int16 i= qweqqw;
```
I think that should not happen.

> Which one of the LocalVariableDeclarations do you what to see? Only the one for the closest ancestor StatementList?
Yes, I think so. In general it is a cool feature to also show the other things, but currently it is not visible from the menu where it will be created

closed 2016-01-21 by slisson