mbeddr/mbeddr.core #1284: buildLanguages script requires write permissions

wuryel, 2016-01-25:
The buildLanguages.sh script requires write permissions in the directory MPS is installed in.
(I'm working in a Linux environment where MPS is installed globally and the mbeddr repository is cloned in the users home directory.)

* log files are written to .../MPS/system/log
* plugins are installed in .../MPS/plugins

It must be ensured that the user executing the script has sufficient rights on those directories.

Please augment the respective section in http://mbeddr.com/download.html (or change the matter)

coolya, 2016-01-28:
That logs files are written pretty much looks like a bug to me. We will investigate what that is the case.

Installing theses plugins on every build is not really necessary it is only need for bootstrapping when MPS is updated or mbeddr is installed the first time. We should split that into to scripts one that *bootstraps* (includes building mbeddr) and one that only builds mbeddr with no write access.

alexanderpann, 2024-11-25:
This script doesn't exist anymore.

closed 2024-11-25 by alexanderpann