mbeddr/mbeddr.core #1319: makefile generation

wuryel, 2016-03-01:
For my application I needed to fix/augment the makefile generation at a few points. I'm engaging the gcc (and a cross-compiler variant of the gcc) in a linux environment.
1) multiple mbeddr libraries need to be included in a single BuildConfiguration
2) don't compile sources that are part of a library (setting of _OBJ in the makefile)
3) the linker might need some specific options
4) a dedicated archiver should be used in the makefile (e.g. for cross-compiling)

Simple fixes and augmentations are suggested in  #1316

coolya, 2016-03-02:
I had a look at your implementation. 1 and 2 look fine to me. That this was not working was clearly a bug.

For 3 and 4 there are two things. First there is no migration that sets the property values of already existing platform instances. So all existing instance will now show errors. Second is I'm would prefer having theses flags in special kind of platform. There is also a little overlap with some efforts where want to separate the compiler from the platform. So that different compilers can be used easily. And generators can have dependency on compiler features.

I would suggest adding your additional properties to a new platform concept that extends the desktop platform. That way you can avoid the migration issue and still reuse most of the code. Also users of the existing gnu platform will ne be confronted with new fields they have to think about.

wuryel, 2016-03-07:
Ok, I get your point. I will change my implementation accordingly.