mbeddr/mbeddr.core #1326: In MPS 3.3.3, this exception occurs in tables

markusvoelter, 2016-03-05:
ailed to create cell for node: [right] DecTab null[9180827081713352667] in test.ts.components.functional.solver@tests using default editor [node r:d7a76695-1c43-44d8-93b8-45ce1979816e(test.ts.components.functional.solver@tests)/9180827081713352667] 
java.lang.RuntimeException: Cycle detected: [EditorCell_Collection, EditorCell_Collection, EditorCell_Collection, EditorCell_Collection, EditorCell_Collection, EditorCell_Collection, EditorCell_Collection] / EditorCell_Collection
	at de.slisson.mps.tables.runtime.cells.TableUtils.checkForCycles(TableUtils.java:134)
	at de.slisson.mps.tables.runtime.cells.TableUtils.checkForCycles(TableUtils.java:140)
	at de.slisson.mps.tables.runtime.cells.TableUtils.checkForCycles(TableUtils.java:129)
	at de.slisson.mps.tables.runtime.cells.TableEditor.init(TableEditor.java:47)
	at org.iets3.core.expr.util.editor.DecTab_Editor$1$1.run(DecTab_Editor.java:102)
	at de.slisson.mps.hacks.editor.EditorCacheHacks.noCaching(EditorCacheHacks.java:14)
	at org.iets3.core.expr.util.editor.DecTab_Editor$1.invoke(DecTab_Editor.java:73)
	at org.iets3.core.expr.util.editor.DecTab_Editor.createTable_6kxgu4_a0(DecTab_Editor.java:112)
	at org.iets3.core.expr.util.editor.DecTab_Editor.createCollection_6kxgu4_a(DecTab_Editor.java:62)
	at org.iets3.core.expr.util.editor.DecTab_Editor.createEditorCell(DecTab_Editor.java:56)
	at jetbrains.mps.nodeEditor.cells.EditorCellFactoryImpl.createCell(EditorCellFactoryImpl.java:135)
	at jetbrains.mps.nodeEditor.cells.EditorCellFactoryImpl.createEditorCell_internal(EditorCellFactoryImpl.java:109)
	at jetbrains.mps.nodeEditor.cells.EditorCellFactoryImpl.createEditorCell(EditorCellFactoryImpl.java:89)
	at com.mbeddr.analyses.cbmc.editor.BaseConcept_conditionalEditor_Editor_VerificationConditions_Coloring.createNextEditor_46qgkg_a0a(BaseConcept_conditionalEditor_Editor_VerificationConditions_Coloring.java:1285)
	at com.mbeddr.analyses.cbmc.editor.BaseConcept_conditionalEditor_Editor_VerificationConditions_Coloring.createAlternation_46qgkg_a0(BaseConcept_conditionalEditor_Editor_VerificationConditions_Coloring.java:68)
	at com.mbeddr.analyses.cbmc.editor.BaseConcept_conditionalEditor_Editor_VerificationConditions_Coloring.createPriorityInfoCell_46qgkg_a(BaseConcept_conditionalEditor_Editor_VerificationConditions_Coloring.java:54)
	at com.mbeddr.analyses.cbmc.editor.BaseConcept_conditionalEditor_Editor_VerificationConditions_Coloring.createEditorCell(BaseConcept_conditionalEditor_Editor_VerificationConditions_Coloring.java:47)
	at jetbrains.mps.nodeEditor.cells.EditorCellFactoryImpl.createCell(EditorCellFactoryImpl.java:135)
	at jetbrains.mps.nodeEditor.cells.EditorCellFactoryImpl.createEditorCell_internal(EditorCellFactoryImpl.java:109)
	at jetbrains.mps.nodeEditor.cells.EditorCellFactoryImpl.createEditorCell(EditorCellFactoryImpl.java:89)
	at com.mbeddr.analyses.cbmc.editor.BaseConcept_conditionalEditor_Editor_TraceProjectionInCode.createNextEditor_46qgkg_a0a(BaseConcept_conditionalEditor_Editor_TraceProjectionInCode.java:517)
	at com.mbeddr.analyses.cbmc.editor.BaseConcept_conditionalEditor_Editor_TraceProjectionInCode.createAlternation_46qgkg_a0(BaseConcept_conditionalEditor_Editor_TraceProjectionInCode.java:62)
	at com.mbeddr.analyses.cbmc.editor.BaseConcept_conditionalEditor_Editor_TraceProjectionInCode.createPriorityInfoCell_46qgkg_a(BaseConcept_conditionalEditor_Editor_TraceProjectionInCode.java:48)
	at com.mbeddr.analyses.cbmc.editor.BaseConcept_conditionalEditor_Editor_TraceProjectionInCode.createEditorCell(BaseConcept_conditionalEditor_Editor_TraceProjectionInCode.java:41)
	at jetbrains.mps.nodeEditor.cells.EditorCellFactoryImpl.createCell(EditorCellFactoryImpl.java:135)
	at jetbrains.mps.nodeEditor.cells.EditorCellFactoryImpl.createEditorCell_internal(EditorCellFactoryImpl.java:109)
	at jetbrains.mps.nodeEditor.cells.EditorCellFactoryImpl.createEditorCell(EditorCellFactoryImpl.java:89)
	at de.itemis.mps.editor.diagram.layout.editor.BaseConcept_conditionalEditor_Editor_connectionLayoutInInspector.createNextEditor_46qgkg_a0a_0(BaseConcept_conditionalEditor_Editor_connectionLayoutInInspector.java:63)
	at de.itemis.mps.editor.diagram.layout.editor.BaseConcept_conditionalEditor_Editor_connectionLayoutInInspector.createAlternation_46qgkg_a0(BaseConcept_conditionalEditor_Editor_connectionLayoutInInspector.java:48)
	at de.itemis.mps.editor.diagram.layout.editor.BaseConcept_conditionalEditor_Editor_connectionLayoutInInspector.createPriorityInfoCell_46qgkg_a(BaseConcept_conditionalEditor_Editor_connectionLayoutInInspector.java:34)
	at de.itemis.mps.editor.diagram.layout.editor.BaseConcept_conditionalEditor_Editor_connectionLayoutInInspector.createEditorCell(BaseConcept_conditionalEditor_Editor_connectionLayoutInInspector.java:27)
	at jetbrains.mps.nodeEditor.cells.EditorCellFactoryImpl.createCell(EditorCellFactoryImpl.java:135)
	at jetbrains.mps.nodeEditor.cells.EditorCellFactoryImpl.createEditorCell_internal(EditorCellFactoryImpl.java:109)
	at jetbrains.mps.nodeEditor.cells.EditorCellFactoryImpl.createEditorCell(EditorCellFactoryImpl.java:89)
	at de.itemis.mps.editor.diagram.layout.editor.BaseConcept_conditionalEditor_Editor_boxLayoutInInspector.createNextEditor_46qgkg_a0a_0(BaseConcept_conditionalEditor_Editor_boxLayoutInInspector.java:63)
	at de.itemis.mps.editor.diagram.layout.editor.BaseConcept_conditionalEditor_Editor_boxLayoutInInspector.createAlternation_46qgkg_a0(BaseConcept_conditionalEditor_Editor_boxLayoutInInspector.java:48)
	at de.itemis.mps.editor.diagram.layout.editor.BaseConcept_conditionalEditor_Editor_boxLayoutInInspector.createPriorityInfoCell_46qgkg_a(BaseConcept_conditionalEditor_Editor_boxLayoutInInspector.java:34)
	at de.itemis.mps.editor.diagram.layout.editor.BaseConcept_conditionalEditor_Editor_boxLayoutInInspector.createEditorCell(BaseConcept_conditionalEditor_Editor_boxLayoutInInspector.java:27)
	at jetbrains.mps.nodeEditor.cells.EditorCellFactoryImpl.createCell(EditorCellFactoryImpl.java:135)
	at jetbrains.mps.nodeEditor.cells.EditorCellFactoryImpl.createEditorCell_internal(EditorCellFactoryImpl.java:109)
	at jetbrains.mps.nodeEditor.cells.EditorCellFactoryImpl.createEditorCell(EditorCellFactoryImpl.java:99)
	at jetbrains.mps.nodeEditor.EditorManager.createEditorCell_internal(EditorManager.java:457)
	at jetbrains.mps.nodeEditor.EditorManager.createEditorCell(EditorManager.java:348)
	at jetbrains.mps.nodeEditor.EditorManager.createEditorCell(EditorManager.java:295)
	at jetbrains.mps.nodeEditor.updater.UpdateSessionImpl$1.compute(UpdateSessionImpl.java:197)
	at jetbrains.mps.nodeEditor.updater.UpdateSessionImpl$1.compute(UpdateSessionImpl.java:194)
	at jetbrains.mps.nodeEditor.updater.UpdateSessionImpl.runWithExplicitEditorHints(UpdateSessionImpl.java:271)
	at jetbrains.mps.nodeEditor.updater.UpdateSessionImpl.updateChildNodeCell(UpdateSessionImpl.java:194)
	at jetbrains.mps.lang.editor.cellProviders.SingleRoleCellProvider.createChildCell_internal(SingleRoleCellProvider.java:68)
	at jetbrains.mps.lang.editor.cellProviders.SingleRoleCellProvider.createChildCell(SingleRoleCellProvider.java:59)
	at jetbrains.mps.lang.editor.cellProviders.SingleRoleCellProvider.createChildCell(SingleRoleCellProvider.java:63)
	at org.iets3.core.expr.base.editor.BinaryExpression_Editor$rightSingleRoleHandler_uefsp_a2a.createChildCell(BinaryExpression_Editor.java:295)
	at jetbrains.mps.lang.editor.cellProviders.SingleRoleCellProvider.createSingleCell(SingleRoleCellProvider.java:116)
	at jetbrains.mps.lang.editor.cellProviders.SingleRoleCellProvider.createCell(SingleRoleCellProvider.java:82)
	at org.iets3.core.expr.base.editor.BinaryExpression_Editor.createRefNode_uefsp_a2a(BinaryExpression_Editor.java:288)
	at org.iets3.core.expr.base.editor.BinaryExpression_Editor.createCustomFactory_uefsp_c0(BinaryExpression_Editor.java:236)
	at org.iets3.core.expr.base.editor.BinaryExpression_Editor.createCollection_uefsp_a(BinaryExpression_Editor.java:42)
	at org.iets3.core.expr.base.editor.BinaryExpression_Editor.createEditorCell(BinaryExpression_Editor.java:34)
	at jetbrains.mps.nodeEditor.cells.EditorCellFactoryImpl.createCell(EditorCellFactoryImpl.java:135)
	at jetbrains.mps.nodeEditor.cells.EditorCellFactoryImpl.createEditorCell_internal(EditorCellFactoryImpl.java:109)
	at jetbrains.mps.nodeEditor.cells.EditorCellFactoryImpl.createEditorCell(EditorCellFactoryImpl.java:89)
[... cut ...]
	at jetbrains.mps.nodeEditor.updater.UpdateSessionImpl$1.compute(UpdateSessionImpl.java:197)
	at jetbrains.mps.nodeEditor.updater.UpdateSessionImpl$1.compute(UpdateSessionImpl.java:194)
	at jetbrains.mps.nodeEditor.updater.UpdateSessionImpl.runWithExplicitEditorHints(UpdateSessionImpl.java:271)
	at jetbrains.mps.nodeEditor.updater.UpdateSessionImpl.updateChildNodeCell(UpdateSessionImpl.java:194)
	at jetbrains.mps.lang.editor.cellProviders.RefNodeListHandler.createNodeCell(RefNodeListHandler.java:80)

slisson, 2016-03-05:
I changed the error message contain more useful information. Can you rebuild the tables.runtime and post the new exception?

markusvoelter, 2016-03-05:
Here you go :-)

[ 286050]   WARN - or.cells.EditorCellFactoryImpl - Failed to create cell for node: [right] DecTab null[7293299957843810903] in sandbox.components.m1 using default editor [node r:1c91eca5-c732-4215-a51f-fc16a38ce9a4(sandbox.components.m1)/7293299957843810903] 
java.lang.RuntimeException: Cycle detected: [[EditorCell_Collection; Collection_v8lgg0_a; DecTabContent; org.iets3.core.expr.util.structure.DecTabContent], [EditorCell_Collection; Collection_v8lgg0_a; DecTabContent; org.iets3.core.expr.util.structure.DecTabContent], [EditorCell_Collection; Collection_v8lgg0_a; DecTabContent; org.iets3.core.expr.util.structure.DecTabContent], [EditorCell_Collection; Collection_v8lgg0_a; DecTabContent; org.iets3.core.expr.util.structure.DecTabContent], [EditorCell_Collection; Collection_v8lgg0_a; DecTabContent; org.iets3.core.expr.util.structure.DecTabContent], [EditorCell_Collection; Collection_v8lgg0_a; DecTabContent; org.iets3.core.expr.util.structure.DecTabContent], [EditorCell_Collection; Collection_v8lgg0_a; DecTabContent; org.iets3.core.expr.util.structure.DecTabContent], [EditorCell_Collection; Collection_v8lgg0_a; DecTabContent; org.iets3.core.expr.util.structure.DecTabContent], [EditorCell_Collection; Collection_v8lgg0_a; DecTabContent; org.iets3.core.expr.util.structure.DecTabContent], [EditorCell_Collection; Collection_v8lgg0_a; DecTabColHeader; org.iets3.core.expr.util.structure.DecTabColHeader], [EditorCell_Collection; Collection_v8lgg0_a; DecTabColHeader; org.iets3.core.expr.util.structure.DecTabColHeader], [EditorCell_Collection; Collection_v8lgg0_a; DecTabColHeader; org.iets3.core.expr.util.structure.DecTabColHeader], [EditorCell_Collection; Collection_v8lgg0_a; DecTabRowHeader; org.iets3.core.expr.util.structure.DecTabRowHeader]] / [EditorCell_Collection; Collection_v8lgg0_a; DecTabContent; org.iets3.core.expr.util.structure.DecTabContent]
	at de.slisson.mps.tables.runtime.cells.TableUtils.checkForCycles(TableUtils.java:152)
	at de.slisson.mps.tables.runtime.cells.TableUtils.checkForCycles(TableUtils.java:158)
	at de.slisson.mps.tables.runtime.cells.TableUtils.checkForCycles(TableUtils.java:132)
	at de.slisson.mps.tables.runtime.cells.TableEditor.init(TableEditor.java:47)
	at org.iets3.core.expr.util.editor.DecTab_Editor$1$1.run(DecTab_Editor.java:102)
	at de.slisson.mps.hacks.editor.EditorCacheHacks.noCaching(EditorCacheHacks.java:14)
	at org.iets3.core.expr.util.editor.DecTab_Editor$1.invoke(DecTab_Editor.java:73)
	at org.iets3.core.expr.util.editor.DecTab_Editor.createTable_6kxgu4_a0(DecTab_Editor.java:112)
	at org.iets3.core.expr.util.editor.DecTab_Editor.createCollection_6kxgu4_a(DecTab_Editor.java:62)
	at org.iets3.core.expr.util.editor.DecTab_Editor.createEditorCell(DecTab_Editor.java:56)
	at jetbrains.mps.nodeEditor.cells.EditorCellFactoryImpl.createCell(EditorCellFactoryImpl.java:135)
	at jetbrains.mps.nodeEditor.cells.EditorCellFactoryImpl.createEditorCell_internal(EditorCellFactoryImpl.java:109)
	at jetbrains.mps.nodeEditor.cells.EditorCellFactoryImpl.createEditorCell(EditorCellFactoryImpl.java:89)
	at com.mbeddr.analyses.cbmc.editor.BaseConcept_conditionalEditor_Editor_VerificationConditions_Coloring.createNextEditor_46qgkg_a0a(BaseConcept_conditionalEditor_Editor_VerificationConditions_Coloring.java:1285)
	at com.mbeddr.analyses.cbmc.editor.BaseConcept_conditionalEditor_Editor_VerificationConditions_Coloring.createAlternation_46qgkg_a0(BaseConcept_conditionalEditor_Editor_VerificationConditions_Coloring.java:68)
	at com.mbeddr.analyses.cbmc.editor.BaseConcept_conditionalEditor_Editor_VerificationConditions_Coloring.createPriorityInfoCell_46qgkg_a(BaseConcept_conditionalEditor_Editor_VerificationConditions_Coloring.java:54)
	at com.mbeddr.analyses.cbmc.editor.BaseConcept_conditionalEditor_Editor_VerificationConditions_Coloring.createEditorCell(BaseConcept_conditionalEditor_Editor_VerificationConditions_Coloring.java:47)
	at jetbrains.mps.nodeEditor.cells.EditorCellFactoryImpl.createCell(EditorCellFactoryImpl.java:135)
	at jetbrains.mps.nodeEditor.cells.EditorCellFactoryImpl.createEditorCell_internal(EditorCellFactoryImpl.java:109)
	at jetbrains.mps.nodeEditor.cells.EditorCellFactoryImpl.createEditorCell(EditorCellFactoryImpl.java:89)
	at com.mbeddr.analyses.cbmc.editor.BaseConcept_conditionalEditor_Editor_TraceProjectionInCode.createNextEditor_46qgkg_a0a(BaseConcept_conditionalEditor_Editor_TraceProjectionInCode.java:517)
	at com.mbeddr.analyses.cbmc.editor.BaseConcept_conditionalEditor_Editor_TraceProjectionInCode.createAlternation_46qgkg_a0(BaseConcept_conditionalEditor_Editor_TraceProjectionInCode.java:62)
	at com.mbeddr.analyses.cbmc.editor.BaseConcept_conditionalEditor_Editor_TraceProjectionInCode.createPriorityInfoCell_46qgkg_a(BaseConcept_conditionalEditor_Editor_TraceProjectionInCode.java:48)
	at com.mbeddr.analyses.cbmc.editor.BaseConcept_conditionalEditor_Editor_TraceProjectionInCode.createEditorCell(BaseConcept_conditionalEditor_Editor_TraceProjectionInCode.java:41)
	at jetbrains.mps.nodeEditor.cells.EditorCellFactoryImpl.createCell(EditorCellFactoryImpl.java:135)

commit e435fbac (2016-03-06): tables: Fixed #1326 (Cycle detected)
closed 2016-03-06 by slisson