mbeddr/mbeddr.core #1331: Doc aspect for concept links

szabta89, 2016-03-22:
The documentation language allows to annotate certain parts of a documentation with a concept which makes it possible to browse the documentation for a concrete instance in the Documentation view. Add an extension to the doc language to allow annotating with links (property or children) of concepts so that when the user selects a concrete instance of the link for a node, the documentation specific to the link is shown in the view.

wsafonov, 2016-04-07:
The support for documenting concept properties has been added. As for references, currently we assume that when a reference is selected, the documentation for the target element should be shown, e.g. like with a method reference. We may reimplement this to only use the target element in case of smart references and look for the reference documentation in other cases. If such documentation is not found, we may still look for the referenced element documentation instead.

alexanderpann, 2023-02-16:
I am closing because this was implemented a long time ago.

closed 2023-02-16 by alexanderpann