mbeddr/mbeddr.core #1333: Investigate installation problems

coolya, 2016-04-01:
On windows machines make fails because of rm not working.

stephaneberle9, 2016-04-02:
The rm command is included in the MinGW distribution. It is located at C:\mingw\msys\1.0\bin. So Maybe you just have to make sure that the build process is able to "see" it. One way to achieve that would be to let the installer add above the mentioned path (and ideally also C:\mingw\bin if not already done) to the %PATH% environment variable.

stephaneberle9, 2016-04-03:
Meanwhile, I've managed to get a fully functional mbeddr setup under Windows using the installer. And, I guess that I've understood what the problem behind this kind of error:

rm -rf .\bin 
Makefile:21:recipe for target 'clean' failed
process_begin: CreateProcess(NULL, rm -rf ./bin, ...) failed
make (e=2): The system cannot find the file specified

To enable MPS to find make and make to find rm, their installation respective installation locations (C:\mingw\bin and C:\mingw\msys\1.0\bin) must be present on the  %PATH% environment variable. The installer already does take care of that. But there is one somewhat tricky pitfall: At the end of the installation process, the installer invites the user to start mbeddr right away. When doing so, mbeddr is spawned from the installer process and inherits the system environment from the same. It is true that the installer has extended the %PATH% environment variable, however this change becomes not immediately effective but only in processes that started SUBSEQUENTLY BY THE USER (i.e., not spawned by the installer). 

I've double-checked this effect by letting the installer proceed to the first time launch of the newly created mbeddr setup. Then I opened a terminal window from within mbeddr (Ctrl+Alt+T) and entered the command: set path. The resulting value did NOT contain the additional MinGW bin paths. Attempts to build the Hello World application from the mbeddr tutorial ended up in errors similar to the one mentioned above. Next, I shut down mbeddr, restarted it, and repeated the same exercises. This time, I got the enhanced %PATH% environment variable value, and the build of the Hello World app succeeded.

The short conclusion of the long story is, that the installer should either be modified so that mbeddr - if the user decides to launch it at the end of the installation process - is launched in a new process that does not inherit its environment from the installer process (don't know exactly if this is possible, but there should be chances at least). Another - less convenient - possibility would consist in removing the option to launch mbeddr at the end of the installation process so that the user has no choice but to launch it manually and with an environment that is initialized as needed.

coolya, 2016-07-21:
Done.

closed 2016-07-21 by coolya