mbeddr/mbeddr.core #1339: "actionsfilter" cannot be used through the mbeddr-platform binary plugins

danielratiu, 2016-04-08:
I am using MPS 3.3.4 with mbeddr-platform plugins installed. When MPS starts I get the following warnings:

[  16936]   WARN - lassloading.ClassLoadersHolder - Facet of the module com.mbeddr.mpsutil.actionsfilter.runtime [solution] is not valid -- possibly the provide
d idea plugin (in the properties dialog/idea plugin facet tab) cannot be found among the bundled plugins
[  42645]   WARN - nplugins.BaseApplicationPlugin - Destination group was not found id: Internal. Trying to insert jetbrains.mps.ide.actions.ToolsInternal_ActionGroup

When I want to define ActionsFilters I get many ModuleClassNotFound exceptions like the following one:

[  51701]  ERROR - om.BaseCustomApplicationPlugin -
java.lang.NoClassDefFoundError: com/mbeddr/mpsutil/actionsfilter/runtime/ActionsApplicationComponent
        at com.siemens.ct.scanner.projectView.pluginSolution.plugin.ActionModificationsApplicationPlugin_CustomApplicationPlugin.doInit(ActionModificationsAppli
cationPlugin_CustomApplicationPlugin.java:23)
        at jetbrains.mps.plugins.custom.BaseCustomApplicationPlugin.init(BaseCustomApplicationPlugin.java:26)
        at com.siemens.ct.scanner.projectView.pluginSolution.plugin.PluginSolution_ApplicationPlugin.initCustomParts(PluginSolution_ApplicationPlugin.java:32)
        at jetbrains.mps.plugins.applicationplugins.BaseApplicationPlugin.createCustomParts(BaseApplicationPlugin.java:131)
        at jetbrains.mps.plugins.applicationplugins.ApplicationPluginManager.createCustomParts(ApplicationPluginManager.java:113)
        at jetbrains.mps.plugins.applicationplugins.ApplicationPluginManager.afterPluginsCreated(ApplicationPluginManager.java:75)
        at jetbrains.mps.plugins.BasePluginManager.loadPlugins(BasePluginManager.java:76)
        at jetbrains.mps.plugins.PluginLoaderRegistry$2.run(PluginLoaderRegistry.java:130)
        at jetbrains.mps.smodel.WorkbenchModelAccess$6.compute(WorkbenchModelAccess.java:287)
        at jetbrains.mps.smodel.WorkbenchModelAccess$6.compute(WorkbenchModelAccess.java:284)
        at jetbrains.mps.util.ComputeRunnable.compute(ComputeRunnable.java:32)
        at jetbrains.mps.util.AbstractComputeRunnable.run(AbstractComputeRunnable.java:27)
        at jetbrains.mps.smodel.WriteActionDispatcher.run(WriteActionDispatcher.java:39)
        at jetbrains.mps.smodel.WriteActionDispatcher.compute(WriteActionDispatcher.java:49)
        at jetbrains.mps.smodel.WorkbenchModelAccess$7.compute(WorkbenchModelAccess.java:310)
        at jetbrains.mps.smodel.WorkbenchModelAccess$TryWriteActionComputable$1.compute(WorkbenchModelAccess.java:706)
        at com.intellij.openapi.application.impl.ApplicationImpl.runWriteAction(ApplicationImpl.java:1014)
        at jetbrains.mps.smodel.WorkbenchModelAccess$TryWriteActionComputable.compute(WorkbenchModelAccess.java:702)
        at jetbrains.mps.smodel.WorkbenchModelAccess.tryWrite(WorkbenchModelAccess.java:324)
        at jetbrains.mps.smodel.WorkbenchModelAccess.tryWrite(WorkbenchModelAccess.java:291)
        at jetbrains.mps.smodel.EDTExecutor$2.tryRun(EDTExecutor.java:78)
        at jetbrains.mps.smodel.EDTExecutor$Executor.worker(EDTExecutor.java:193)
        at jetbrains.mps.smodel.EDTExecutor$Executor.access$300(EDTExecutor.java:129)
        at jetbrains.mps.smodel.EDTExecutor$Executor$1.run(EDTExecutor.java:135)
        at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
        at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
        at java.awt.EventQueue.access$500(EventQueue.java:97)
        at java.awt.EventQueue$3.run(EventQueue.java:709)
        at java.awt.EventQueue$3.run(EventQueue.java:703)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
        at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
        at com.intellij.ide.IdeEventQueue.defaultDispatchEvent(IdeEventQueue.java:866)
        at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:654)
        at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:381)
        at com.intellij.ide.IdeEventQueue.pumpEventsForHierarchy(IdeEventQueue.java:953)
        at com.intellij.openapi.progress.util.ProgressWindow.startBlocking(ProgressWindow.java:210)
        at com.intellij.openapi.progress.util.ProgressWindow.startBlocking(ProgressWindow.java:197)
        at com.intellij.openapi.application.impl.ApplicationImpl.runProcessWithProgressSynchronously(ApplicationImpl.java:642)
        at com.intellij.openapi.application.impl.ApplicationImpl.runProcessWithProgressSynchronously(ApplicationImpl.java:583)
        at com.intellij.openapi.application.impl.ApplicationImpl.runProcessWithProgressSynchronously(ApplicationImpl.java:574)
        at jetbrains.mps.ide.make.StartupModuleMakerImpl.projectOpened(StartupModuleMakerImpl.java:67)
        at com.intellij.openapi.project.impl.ProjectImpl.projectOpened(ProjectImpl.java:400)
        at com.intellij.openapi.project.impl.ProjectImpl.access$200(ProjectImpl.java:63)
        at com.intellij.openapi.project.impl.ProjectImpl$MyProjectManagerListener.projectOpened(ProjectImpl.java:434)
        at com.intellij.openapi.project.impl.ProjectManagerImpl$1.projectOpened(ProjectManagerImpl.java:98)
        at com.intellij.openapi.project.impl.ProjectManagerImpl.fireProjectOpened(ProjectManagerImpl.java:721)
        at com.intellij.openapi.project.impl.ProjectManagerImpl.openProject(ProjectManagerImpl.java:352)
        at com.intellij.openapi.project.impl.ProjectManagerImpl.loadAndOpenProject(ProjectManagerImpl.java:467)
        at com.intellij.ide.impl.ProjectUtil.openProject(ProjectUtil.java:182)
        at jetbrains.mps.workbench.recent.RecentMPSProjectsManager.doOpenProject(RecentMPSProjectsManager.java:54)
        at com.intellij.ide.RecentProjectsManagerBase.doReopenLastProject(RecentProjectsManagerBase.java:554)
        at com.intellij.ide.RecentProjectsManagerBase$MyAppLifecycleListener.appStarting(RecentProjectsManagerBase.java:588)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.lang.reflect.Method.invoke(Method.java:497)
        at com.intellij.util.messages.impl.MessageBusConnectionImpl.deliverMessage(MessageBusConnectionImpl.java:116)
        at com.intellij.util.messages.impl.MessageBusImpl.doPumpMessages(MessageBusImpl.java:368)
[... 7 lines cut ...]
        at com.intellij.openapi.application.impl.LaterInvocator$FlushQueue.run(LaterInvocator.java:321)
        at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
        at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
        at java.awt.EventQueue.access$500(EventQueue.java:97)
        at java.awt.EventQueue$3.run(EventQueue.java:709)
        at java.awt.EventQueue$3.run(EventQueue.java:703)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
        at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
        at com.intellij.ide.IdeEventQueue.defaultDispatchEvent(IdeEventQueue.java:866)
        at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:654)
        at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:381)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
        at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
        at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
Caused by: jetbrains.mps.classloading.ModuleClassNotFoundException: Unable to load class: com.mbeddr.mpsutil.actionsfilter.runtime.ActionsApplicationComponent u
sing ModuleClassLoader of com.siemens.ct.scanner.projectView.pluginSolution module
        at jetbrains.mps.classloading.ModuleClassLoader.createCLNFException(ModuleClassLoader.java:136)
        at jetbrains.mps.classloading.ModuleClassLoader.loadClass(ModuleClassLoader.java:123)
        at jetbrains.mps.classloading.ModuleClassLoader.loadClass(ModuleClassLoader.java:90)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
        ... 84 more

danielratiu, 2016-04-08:
in the build.xml, as part of build-platform task, we copy the actionsfilter into plugins directory of the running MPS instance

<target name="build-platform">
		<ant antfile="${com.mbeddr.platform.build-file.name}" dir="${com.mbeddr.platform.build-file.location}" target="${generate-code.task}" />
		<ant antfile="${com.mbeddr.platform.build-file.name}" dir="${com.mbeddr.platform.build-file.location}" target="${build-plugin.task}" />   
		<copy todir="${mps.home}/${plugins.folder.name}">
			<zipfileset src="${artifacts.root}/com.mbeddr.platform/com.mbeddr.mpsutil.actionsfilter.zip">
			</zipfileset>
		</copy>
		<antcall target="build-spawner" />
	</target> 

By manually copying the com.mbeddr.mpsutil.actionsfilter into the MPS instance where binary plugins are installed, the problem is away.

DomenikP, 2016-04-08:
Ok, commit that to master!.

danielratiu, 2016-04-08:
Domenik, I don't know what to commit ...

coolya, 2016-07-21:
seems fixed to me, its included in the distribution.

closed 2016-07-21 by coolya