mbeddr/mbeddr.core #1341: java.lang.NullPointerException: open empty JNGraph with ISOM layout

KeSUN-X, 2016-04-11:
When an empty JNGraph (no vertex in it) is opened with ISOM layout, a NullPointerException will be thrown.

markusvoelter, 2016-04-12:
How do you configure PlantUML to use ISOM layout? Or is this the default? And can you please post the stack trace of the exception?

KeSUN-X, 2016-04-12:
Hello Markus,

I just followed DepGraphHelper.

public static JNGraph createGraph (string graphName) {
    final JNGraph graph=newJNGraph(graphName, newJNISOMLayout(),
newJNHighlightMode(true,true,false,false), 600);
    ...
    graph;
}

When no vertex is created in JNGraph (graph.createVertex(...) is not
invoked), the command "show graph viewer" will report an error. On the
other hand, if the graph is initialized with another layout, the command
works well.

[ 663794]  ERROR - .workbench.action.BaseAction$2 - User's action execute
method failed. Action: Show Graph Viewer. Class:
com.mbeddr.mpsutil.jung.pluginSolution.plugin.OpenGraphViewer_Action
java.lang.NullPointerException
        at
edu.uci.ics.jung.algorithms.layout.ISOMLayout.adjustVertex(ISOMLayout.java:176)
        at
edu.uci.ics.jung.algorithms.layout.ISOMLayout.adjust(ISOMLayout.java:139)
        at
edu.uci.ics.jung.algorithms.layout.ISOMLayout.step(ISOMLayout.java:107)
        at
edu.uci.ics.jung.algorithms.layout.LayoutDecorator.step(LayoutDecorator.java:58)
        at
edu.uci.ics.jung.visualization.layout.ObservableCachingLayout.step(ObservableCachingLayout.java:59)
        at
edu.uci.ics.jung.algorithms.layout.util.VisRunner.prerelax(VisRunner.java:62)
        at
edu.uci.ics.jung.visualization.DefaultVisualizationModel.setGraphLayout(DefaultVisualizationModel.java:113)
        at
edu.uci.ics.jung.visualization.DefaultVisualizationModel.<init>(DefaultVisualizationModel.java:74)
        at
edu.uci.ics.jung.visualization.DefaultVisualizationModel.<init>(DefaultVisualizationModel.java:58)
        at
edu.uci.ics.jung.visualization.VisualizationViewer.<init>(VisualizationViewer.java:63)
        at
com.mbeddr.mpsutil.jung.pluginSolution.plugin.GraphViewer_Tool.updateGraph(GraphViewer_Tool.java:217)
        at
com.mbeddr.mpsutil.jung.pluginSolution.plugin.GraphViewer_Tool.setNewGraph(GraphViewer_Tool.java:189)
        at
com.mbeddr.mpsutil.jung.pluginSolution.plugin.OpenGraphViewer_Action.doExecute(OpenGraphViewer_Action.java:73)
        at
jetbrains.mps.workbench.action.BaseAction$2.run(BaseAction.java:145)
        at
jetbrains.mps.smodel.WorkbenchModelAccess$CommandRunnable$1.run(WorkbenchModelAccess.java:642)
        at
jetbrains.mps.smodel.WriteActionDispatcher.run(WriteActionDispatcher.java:39)
        at
jetbrains.mps.smodel.WorkbenchModelAccess$3.run(WorkbenchModelAccess.java:135)
        at
com.intellij.openapi.application.impl.ApplicationImpl.runWriteAction(ApplicationImpl.java:1002)
        at
jetbrains.mps.smodel.WorkbenchModelAccess.runWriteAction(WorkbenchModelAccess.java:144)
        at
jetbrains.mps.smodel.WorkbenchModelAccess$CommandRunnable.run(WorkbenchModelAccess.java:637)
        at
com.intellij.openapi.command.impl.CoreCommandProcessor.executeCommand(CoreCommandProcessor.java:124)
        at
com.intellij.openapi.command.impl.CoreCommandProcessor.executeCommand(CoreCommandProcessor.java:99)
        at
jetbrains.mps.smodel.WorkbenchModelAccess.runWriteActionInCommand(WorkbenchModelAccess.java:450)
        at
jetbrains.mps.smodel.WorkbenchModelAccess.executeCommand(WorkbenchModelAccess.java:413)
        at
jetbrains.mps.project.ProjectModelAccess.executeCommand(ProjectModelAccess.java:43)
        at
jetbrains.mps.workbench.action.BaseAction.actionPerformed(BaseAction.java:160)
        at
com.intellij.openapi.keymap.impl.IdeKeyEventDispatcher$3.performAction(IdeKeyEventDispatcher.java:595)
        at
com.intellij.openapi.keymap.impl.IdeKeyEventDispatcher.processAction(IdeKeyEventDispatcher.java:646)
        at
com.intellij.openapi.keymap.impl.IdeKeyEventDispatcher.inInitState(IdeKeyEventDispatcher.java:485)
        at
com.intellij.openapi.keymap.impl.IdeKeyEventDispatcher.dispatchKeyEvent(IdeKeyEventDispatcher.java:215)
        at
com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:622)
        at
com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:381)
        at
java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:242)
        at
java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:161)
        at
java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:150)
        at
java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:146)
        at
java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:138)
        at java.awt.EventDispatchThread.run(EventDispatchThread.java:91)

Best regards, Cordialement, Viele Grüsse

Ke SUN
----------------------------------------
Software Engineer
itemis France SAS
6, Rue Dewoitine
Immeuble SAPHIR
78140 Vélizy-Villacoublay
France

Mobile: +33 (0)6 67 63 26 60 (preferred)
Phone: +33 (0)9 86 35 99 25
Web: http://www.itemis.com
Mail: sun@itemis.com
----------------------------------------

markusvoelter, 2016-04-12:
The NPE is not easily detectable, but I have added this code at the beginning of update().

if(gr.vertices().isEmpty){ 
  System.err.println("JUNG Graph is empty (no vertices); skipping update."); 
  return; 
}

does this help?

closed 2016-04-12 by markusvoelter