mbeddr/mbeddr.core #1352: "java.lang.IllegalArgumentException: Not a IBoxAccessor" when using diagram edge editor

levilucio, 2016-04-15:
When I made my simple state machine language I used the code from the mbeddr StateMachines  to graphically display the edges. I got this exception when rendering the diagram after no compilation errors:

java.lang.IllegalArgumentException: Not a IBoxAccessor: org.iets3.core.simplestatemachine.editor.Transition_SSMAsADiagram_Editor$1$1
	at de.itemis.mps.editor.diagram.runtime.model.Box.setAccessor(Box.java:142)
	at de.itemis.mps.editor.diagram.runtime.model.ModelSynchronizer.fromSModel_(ModelSynchronizer.java:242)
	at de.itemis.mps.editor.diagram.runtime.model.ModelSynchronizer$7$1.run(ModelSynchronizer.java:186)
	at de.itemis.mps.editor.diagram.runtime.model.ScalableEditorCell.withScalingDisabled(ScalableEditorCell.java:27)
	at de.itemis.mps.editor.diagram.runtime.model.ModelSynchronizer$7.run(ModelSynchronizer.java:184)
	at jetbrains.mps.smodel.WorkbenchModelAccess.runReadAction(WorkbenchModelAccess.java:96)
	at jetbrains.mps.smodel.ModelAccessBase.runReadAction(ModelAccessBase.java:53)
	at de.itemis.mps.editor.diagram.runtime.model.ModelSynchronizer.fromSModel(ModelSynchronizer.java:182)
	at de.itemis.mps.editor.diagram.runtime.model.DiagramModel$2.run(DiagramModel.java:222)
	at de.itemis.mps.editor.diagram.runtime.model.ScalableEditorCell.withScalingDisabled(ScalableEditorCell.java:27)
	at de.itemis.mps.editor.diagram.runtime.model.DiagramModel.synchronizeRead(DiagramModel.java:220)
	at de.itemis.mps.editor.diagram.runtime.jgraph.RootDiagramECell$4.run(RootDiagramECell.java:135)
	at de.itemis.mps.editor.diagram.runtime.ContextVariables$1.run(ContextVariables.java:40)
	at de.itemis.mps.editor.diagram.runtime.ContextVariables.enterNewContext(ContextVariables.java:30)
	at de.itemis.mps.editor.diagram.runtime.ContextVariables.withValue(ContextVariables.java:37)
	at de.itemis.mps.editor.diagram.runtime.jgraph.BaseDiagramECell.withContextGraph(BaseDiagramECell.java:123)
	at de.itemis.mps.editor.diagram.runtime.jgraph.RootDiagramECell.<init>(RootDiagramECell.java:133)
	at org.iets3.core.simplestatemachine.editor.SimpleStateMachine_SSMAsADiagram_Editor$1$1$1.run(SimpleStateMachine_SSMAsADiagram_Editor.java:285)
	at de.itemis.mps.editor.diagram.runtime.ContextVariables$1.run(ContextVariables.java:40)
	at de.itemis.mps.editor.diagram.runtime.ContextVariables.enterNewContext(ContextVariables.java:30)
	at de.itemis.mps.editor.diagram.runtime.ContextVariables.withValue(ContextVariables.java:37)
	at org.iets3.core.simplestatemachine.editor.SimpleStateMachine_SSMAsADiagram_Editor$1$1.invoke(SimpleStateMachine_SSMAsADiagram_Editor.java:132)
	at de.itemis.mps.editor.diagram.runtime.jgraph.DiagramCreationContext.createDiagram_(DiagramCreationContext.java:51)
	at de.itemis.mps.editor.diagram.runtime.jgraph.DiagramCreationContext$1.run(DiagramCreationContext.java:29)
	at de.itemis.mps.editor.collapsible.runtime.CollapsibleContext.withFactory(CollapsibleContext.java:21)
	at de.itemis.mps.editor.diagram.runtime.jgraph.DiagramCreationContext.createDiagram(DiagramCreationContext.java:27)
	at org.iets3.core.simplestatemachine.editor.SimpleStateMachine_SSMAsADiagram_Editor$1.run(SimpleStateMachine_SSMAsADiagram_Editor.java:130)
	at de.itemis.mps.editor.diagram.runtime.EditorUtil.noCaching(EditorUtil.java:137)
	at org.iets3.core.simplestatemachine.editor.SimpleStateMachine_SSMAsADiagram_Editor.createDiagram_vs9s9l_d0(SimpleStateMachine_SSMAsADiagram_Editor.java:128)
	at org.iets3.core.simplestatemachine.editor.SimpleStateMachine_SSMAsADiagram_Editor.createCollection_vs9s9l_a(SimpleStateMachine_SSMAsADiagram_Editor.java:84)
	at org.iets3.core.simplestatemachine.editor.SimpleStateMachine_SSMAsADiagram_Editor.createEditorCell(SimpleStateMachine_SSMAsADiagram_Editor.java:74)
	at jetbrains.mps.nodeEditor.cells.EditorCellFactoryImpl.createCell(EditorCellFactoryImpl.java:135)
	at jetbrains.mps.nodeEditor.cells.EditorCellFactoryImpl.createEditorCell_internal(EditorCellFactoryImpl.java:109)
	at jetbrains.mps.nodeEditor.cells.EditorCellFactoryImpl.createEditorCell(EditorCellFactoryImpl.java:99)
	at jetbrains.mps.nodeEditor.EditorManager.createEditorCell_internal(EditorManager.java:457)
	at jetbrains.mps.nodeEditor.EditorManager.createEditorCell(EditorManager.java:348)
	at jetbrains.mps.nodeEditor.EditorManager.createEditorCell(EditorManager.java:295)
	at jetbrains.mps.nodeEditor.EditorManager.createRootCell(EditorManager.java:98)
	at jetbrains.mps.nodeEditor.updater.UpdateSessionImpl.performUpdate(UpdateSessionImpl.java:147)
	at jetbrains.mps.nodeEditor.updater.UpdaterImpl$1.compute(UpdaterImpl.java:134)
	at jetbrains.mps.nodeEditor.updater.UpdaterImpl$1.compute(UpdaterImpl.java:131)
	at jetbrains.mps.typesystem.inference.TypeContextManager$Executor.doExecute(TypeContextManager.java:276)
	at jetbrains.mps.typesystem.inference.TypeContextManager$Executor.execute(TypeContextManager.java:246)
	at jetbrains.mps.typesystem.inference.TypeContextManager.runTypecheckingAction(TypeContextManager.java:136)
	at jetbrains.mps.nodeEditor.updater.UpdaterImpl.updateRootCell(UpdaterImpl.java:130)
	at jetbrains.mps.nodeEditor.updater.UpdaterImpl.doUpdate(UpdaterImpl.java:109)
	at jetbrains.mps.nodeEditor.updater.UpdaterImpl.update(UpdaterImpl.java:87)
	at jetbrains.mps.nodeEditor.EditorComponent.rebuildEditorContent(EditorComponent.java:1949)
	at jetbrains.mps.nodeEditor.NodeEditorComponent.rebuildEditorContent(NodeEditorComponent.java:110)
	at jetbrains.mps.nodeEditor.EditorComponent$3$1.run(EditorComponent.java:294)
	at jetbrains.mps.smodel.WorkbenchModelAccess$5.compute(WorkbenchModelAccess.java:225)
	at jetbrains.mps.smodel.WorkbenchModelAccess$5.compute(WorkbenchModelAccess.java:220)
	at com.intellij.openapi.application.impl.ApplicationImpl.runReadAction(ApplicationImpl.java:955)
	at jetbrains.mps.smodel.WorkbenchModelAccess.tryRead(WorkbenchModelAccess.java:220)
	at jetbrains.mps.smodel.EDTExecutor$1.tryRun(EDTExecutor.java:59)
	at jetbrains.mps.smodel.EDTExecutor$Executor.worker(EDTExecutor.java:193)
	at jetbrains.mps.smodel.EDTExecutor$Executor.access$300(EDTExecutor.java:129)
	at jetbrains.mps.smodel.EDTExecutor$Executor$1.run(EDTExecutor.java:135)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at com.intellij.ide.IdeEventQueue.defaultDispatchEvent(IdeEventQueue.java:866)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:654)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:381)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

slisson, 2016-04-21:
The language "org.iets3.core.simplestatemachine" is missing in the repository.

markusvoelter, 2016-04-21:
where is this used? AFAIR, the state machines are in the componetns.functional language

bkolb, 2016-04-21:
This is Levis playground afaik...
Bernd

levilucio, 2016-04-21:
The language is on my branch:

levi_experiments

It is something I was experimenting with and got this exception when displaying edges, although there were no compilation errors. Couldn't understand what I did wrong.

slisson, 2016-04-21:
I found the branch, but you never commited the language.

levilucio, 2016-04-23:
I had to change things around a bit. The language is now on this branch:

traffic_light_for_ZF_stable

closed 2016-04-26 by slisson