mbeddr/mbeddr.core #1356: Generation bug in mbeddr.core in presence of inlined functions

danielratiu, 2016-05-21:
I have two implementation models: m1 contains an inlined function 'boo'; and a module m2 which contains a function 'foo' which calls 'boo'. The module m1 is marked as reexported.

The code generation fails.

[image]

[image]

[image]

[inline_bug.zip](https://github.com/mbeddr/mbeddr.core/files/275841/inline_bug.zip)

danielratiu, 2016-05-21:
here is the error message:

[image]

bkolb, 2016-06-03:
@stephaneberle9 can you please have a look at that? i think you reworked that, right?

coolya, 2016-06-09:
fixed

closed 2016-06-09 by coolya