mbeddr/mbeddr.core #1396: RuntimeException when using File > New > mbeddr Project wizard

stephaneberle9, 2016-06-16:
When using the File > New > mbeddr Project wizard, we end up with an MPS project that is stored in an .mpr file. This seems not to be in sync with the normal MPS project creation policy (File > New > Project) producing MPS projects that are stored in an .mps folder.

In addition (or because of that) we systematically get the following exception when creating a new mbeddr project:

RuntimeException: java.lang.RuntimeException : java.io.IOException: 'C:/Users/steph/MPSProjects/NewProject/NewProject.mpr' is not a directory in VFS: java.io.IOException: 'C:/Users/steph/MPSProjects/NewProject/NewProject.mpr' is not a directory in VFS

slisson, 2016-06-16:
I fixed it already in eb88ef990978ddcae53513e96362ef84ef0ad892

stephaneberle9, 2016-06-16:
Ok, thanks. 

Then this was just a "timing issue" - apparently we have picked up the wrong build for validating the absence of this problem.

closed 2016-06-16 by stephaneberle9