mbeddr/mbeddr.core #1399: rm.exe not part of minGW anymore

DomenikP, 2016-06-16:
The minGW installation does not come with rm.exe anymore, which is required by our generated make files. It seems like they shipped the required binary in previous versions, but removed it now.

stephaneberle9, 2016-06-16:
The GNU ARM Eclipse problem has already been face to the same problem. Maybe we can implement a similar workaround or, even better, just grab the resulting extra pieces of binaries from them.

http://gnuarmeclipse.github.io/windows-build-tools/install/

stephaneberle9, 2016-06-16:
Another potential option would be to switch to tdm-gcc:

http://tdm-gcc.tdragon.net/about

coolya, 2016-07-16:
Worked on windows 7 with the current mingw installer, test on windows 10 pending.

mbeddr-buildbot, 2016-07-16:
Have you installed the same packages that we list on mbeddr.com?

coolya, 2016-07-21:
I used the installer. 

You can see what is installed here: https://github.com/mbeddr/mbeddr.installer.win/blob/master/mbeddr.nsi#L93

closed 2016-07-21 by coolya