mbeddr/mbeddr.core #1420: Running tests from the command line with MPS open cause exception

ftomassetti, 2016-07-07:
The exception is about the address for the internal server being already in use.
Would it be possible to either:
1. Check if the port is free before starting the server
2. Specify as a configuration parameter the port to be used for tests

coolya, 2016-07-20:
Whats the problem here? Do the tests stop because of this? I don't think so. I think there is no quick workaround.

ftomassetti, 2016-07-21:
No, it is not making the tests fail (at least not the one not depending on the internal server). However it is distracting. In general these exceptions are most of time not the symptom of an issue but they could possibly hide an issue. In this sense it would be cool to remove them. For example if running the test I could pass an environment variable which overrides the HTTP port to be used that would solve the issue I think.

coolya, 2016-07-21:
Seems like a good idea. We will introduce  a environment variable that has priority over the settings.

commit 6231d1f0 (2016-07-31): mpsutil: http server takes port from property or env when in unit test

if the application is in unittest mode the http server now picks up the port
it uses from the env variable "MBEDDR_TEST_HTTP_PORT". If that variable is not
present or does not contain a valid port the property mbeddr.test.http.port is read
if it contains a valid TCP port, this port it used.

fixes #1420
PR #1457 (2016-11-09): mpsutil: http server takes port from property or env when in unit test
if the application is in unittest mode the http server now picks up the port
it uses from the env variable "MBEDDR_TEST_HTTP_PORT". If that variable is not
present or does not contain a valid port the property mbeddr.test.http.port is read
if it contains a valid TCP port, this port it used.

fixes #1420
commit a4438335 (2016-11-09): mpsutil: http server takes port from property or env when in unit test

if the application is in unittest mode the http server now picks up the port
it uses from the env variable "MBEDDR_TEST_HTTP_PORT". If that variable is not
present or does not contain a valid port the property mbeddr.test.http.port is read
if it contains a valid TCP port, this port it used.

fixes #1420
closed 2016-11-09 by coolya