mbeddr/mbeddr.core #1453: [exported reexport] The exported property requires imported module to be reexported

ksun-x, 2016-07-27:
For instance, an implementation module `myModule` has an **exported** enum declaration, which uses a constant `myConstant` defined in a **non-reexported** imported module `myLib`. In current version, no error like `must be reexported` is reported. 

After generation, the enum declaration is located in the generated header file `myModule.h` while `#include "myLib.h"` is in `myModule.c`. Hence an error `myConstant undeclared here` would be thrown during make.

ksun-x, 2016-07-28:
One more concrete case:

In the `com.itemis.embedded.platform.raspi.advaboardrpi1.implementation.impl.AbstractDigitalIOPinImpl` module, the `configurePin` runnable has a parameter `ioConfig` whose type is an enum `am_config_io`. This enum is declared in an imported module `am_rpi_advaboard`.

When `am_rpi_advaboard` is **non-reexported**, no error is reported. However, when this module is rebuilt in the model `com.itemis.embedded.platform.examples.towerofhanoi.integrations.advaboardrpi1.win32.io`, an error is reported to indicate that `#include <am_rpi_advaboard.h>` should be located in `AbstractDigitalIOPinImpl.h` rather than in `AbstractDigitalIOPinImpl.c`. 

To avoid this case, `am_rpi_advaboard` should be checked whether it **`must be reexported`**.

alexanderpann, 2022-04-27:
Closing because seems to be fixed in https://github.com/mbeddr/mbeddr.core/pull/1699.

closed 2022-04-27 by alexanderpann