mbeddr/mbeddr.core #1467: Auto layout option moves the contents of boxes (it should not)

joswarmer, 2016-08-04:
First click on "Fit size of All Boxes".   Then click on "Auto Layout Diagram".  The Auto layout action then changes the location of the contents inside the boxes in the diagram.  I was expecting that it would only re-layout the boxes, not their contents.

If I select a box after the re-layout, the box gets selected and its contents immediately moves back to the position after the "Fix Size of All boxes".  My guess is that the move of the contents after the re-layout is probably wrong.

joswarmer, 2016-08-05:
Sometimes, :Fix size of all boxes" does not restore the position of the internal contents of a box,  only selecting the box does.

joswarmer, 2016-08-05:
Ans sometimes selecting does not restore the position,  I first need to drag the box.

joswarmer, 2016-08-05:
Sometimes I just have to select one box and all contents of all boxes in the diagram move to the correct position at once!

slisson, 2016-08-10:
Similar to #1411

closed 2018-08-06 by kbirken