mbeddr/mbeddr.core #1470: Allow nodes to be visible in multiple diagrams

joswarmer, 2016-08-11:
A node is often visible in multiple diagrams. E.g. when inherited nodes are shown somewhere in an extends hierarchy.  Currently you cannot use the _nodesQuery_ to get nodes into more than one diagram,  as the _nodesQuery_ used the same ID for one node to store diagram information.  

The current workaround is to use a _generix box query_  instead of a _nodesQuery_ because the _generic box query_ allows you to specify the ID yourself,  so you can make a unique ID.

However,  the _generic box query_ is much more limited, and forces you to specify many details that are done more or less automatically in case of a _nodesQuery_. One specific example is e.g. that a _generic box query_ forces you to specify one node type only for the result of the query. And then you have to specify properties like the _shape_, the _editor_, etc. explicitly.  The _nodesQuery_  allows you to just return a set of randon nodes and it will automatically find the editor and shape available.

For a _nodesQuery_ that results in many different node types, I now have to filter the query by type, and use a _generic box query_ for each of these types. This blows up the code of the editor a lot.

Therefore it should be allowed to have a node visible in multiple diagrams without having to use the _generic box query_.

slisson, 2016-08-11:
The generic box query is not a workaround, it's the only possible way. This is because of limitations in MPS.

joswarmer, 2016-08-11:
:-(

slisson, 2016-08-11:
I can try to make it more extensible.

joswarmer, 2016-08-11:
Alternatively, would is be possible to make the _generic box query_ more flexible to solve some of the problems?

joswarmer, 2016-08-11:
Just thinking aloud: would it e.g. be possible to let the generic box query refer to the diagram box editor for the parameter node type?  This could result in a lot less duplicated editor specification code, because all of the properties are defined in that editor as well.

slisson, 2016-08-11:
I can try to invoke the cell factory for the concept directly, but I don't know how well that will work and what the side effects are.

joswarmer, 2016-08-11:
Can you allow a reference to the nodes diagram editor from the generic box query,  and then copy that, such that it essentially becomes a short way to specify the same as we have today ?

slisson, 2016-08-11:
This would require to reference all editors for all possible subconcepts and break extensibility.

joswarmer, 2016-08-11:
You are right that this isn't extensible.  But,  in the current generic box query we need to specify everything explicitly for the specified node type, which cannot be overridden by possible subconcepts. Doesn't this mean that this is non-extensible either? Or am I missing some extensibility option here?

slisson, 2016-08-11:
Yes, both options would break extensibility. That's why I'm trying to find a solution that does not require generic queries.

slisson, 2016-08-12:
Changes related to this issue:
d1685402ade6f90048db7863eaf1c6c0dba10193
82c2513660b3e4db8d575b9439d4d28897123b04

joswarmer, 2016-08-15:
Changing the layout only in the second diagram changes the actual model.  In 
[image]
Composite A defines a model, with connections from 'str' to 'amplifier'. In Composite C I see 'amplifier' because it is inherited from 'composite A'.  When I move any element in diagram C,  or use the Fit Size of All Boxes'  option, not only diagram C changes, but the connections in diagram A are deleted from the model:
[image]
I am using the untested "duplicates safe" option in the definition of the diagram.

closed 2016-08-30 by slisson