mbeddr/mbeddr.core #1471: Be able to distinguish in which context a node is displayed in the diagram editor

joswarmer, 2016-08-11:
When issue #1470 is resolved, there is an additional need.

The nodes that are shown e.g. because they are inherited from an extended block should behave (and show) slightly different, e.g. you should not be able to delete nodes that show because they come from an extended block.  
Some information about the context in which a node is shown is needed. E.g. can I get access to the diagram node (i.e. the node that constitutes the full diagram) from an expression inside a diagram editor?

Partial solution (by Sascha): use `DiagramCreationContext.getRootGraph().getRootDiagramModel().getSNode()
`

This seems to works ok, but only when the diagram is created, so I can tweak the projection..  It does not give me the context when a node is selected later on to tweak the behavior.

slisson, 2016-08-12:
In context actions you can get the node of the selected diagram like this:
context.getSelection() as GraphSelection.?getDiagramCell().?getSNode()

joswarmer, 2016-08-15:
in the context actions I am trying to use context.getSelection(), but this is often null, even though I can see a box selected in the diagram and see it in the inspector.

slisson, 2016-08-16:
b11b6424a0047734bc260024c751b256b49eed06

closed 2016-08-16 by slisson