mbeddr/mbeddr.core #1476: Allow multiple lines of code in queriesa in diagram editor definitions

joswarmer, 2016-08-22:
At several places you can specify a query (e.g. for ports in a diagram.box.  These queries to be exactly one line of base language code. For debugging purposed (e.g. prinln(..) ) it would be good te be able to specify multiple lines of code.
Also, fro a consistentvy point of view this should also be done.  In MPS you can specify a query at many places but this always is a function like things where you can have multiple lines of code.

slisson, 2016-08-29:
You can always use closures, if you need multiple lines. Most of the time only a single line is sufficient and using concept functions everywhere would blow up the code and make it harder to read.

joswarmer, 2016-08-30:
Ok,  I can live with that,  as one line will be enough most of the time. Shall we close this issue ?

closed 2016-08-30 by slisson