mbeddr/mbeddr.core #1479: Editor hints inside box not seen

joswarmer, 2016-08-30:
I am using an editor hint (`graphical` from mbeddr state machines) in a textual editor definition. Instances of this concept that appear inside a box in a diagram seem to ignore this hint and keep using the default editor.   I have invalidated caches, rebuilt everything etc, but the problem remains. For some reason the editor hint is not seen by the editor inside the box.

slisson, 2016-08-30:
Where can I see this issue? I don't know how to reproduce it. In my example, the hints are available.

joswarmer, 2016-08-30:
Just pushed my changes to milestone/v15-2.  Just create a State and then add action items (entry, exit, ... inside the state.
These items should use the `graphical` editor of `AbstractAction`, but do follow the `default` editor instead.

slisson, 2016-08-30:
In State_graphical_Editor, the [root collection](http://localhost:8080/select/com.mbeddr.ext/r:775fb93d-1c49-4a5d-832a-a94f24da0fdf/8722490720968616260/) removes the "graphical" hint (see inspector).

closed 2016-08-30 by slisson

joswarmer, 2016-08-30:
Tnx ! Didn't even know that was possible.