mbeddr/mbeddr.core #1489: ConcurrentModificationException with MPS 3.4

danielratiu, 2016-09-14:
java.util.ConcurrentModificationException
        at java.util.HashMap$HashIterator.nextNode(HashMap.java:1429)
        at java.util.HashMap$KeyIterator.next(HashMap.java:1453)
        at jetbrains.mps.smodel.language.LanguageRegistry.getLanguage(LanguageRegistry.java:297)
        at jetbrains.mps.editor.runtime.style.StyleAttributes.getAttribute(StyleAttributes.java:119)
        at de.itemis.mps.editor.celllayout.styles.editor.LayoutStylesUtil.getBottomBorderSize(LayoutStylesUtil.java:46)
        at de.itemis.mps.editor.celllayout.runtime.BorderPainter.paintBorder(BorderPainter.java:74)
        at de.itemis.mps.editor.celllayout.runtime.BorderPainter.paintBorders(BorderPainter.java:59)
        at de.itemis.mps.editor.celllayout.runtime.BorderPainter.paintBorders(BorderPainter.java:65)
        at de.itemis.mps.editor.celllayout.runtime.BorderPainter.paintBorders(BorderPainter.java:65)
        at de.itemis.mps.editor.celllayout.runtime.BorderPainter.paintBorders(BorderPainter.java:65)
        at de.itemis.mps.editor.celllayout.runtime.BorderPainter.paintBorders(BorderPainter.java:65)
        at de.itemis.mps.editor.celllayout.runtime.BorderPainter.paintBorders(BorderPainter.java:65)
        at de.itemis.mps.editor.celllayout.runtime.BorderPainter.paintBorders(BorderPainter.java:65)
        at de.itemis.mps.editor.celllayout.runtime.BorderPainter.paintBorders(BorderPainter.java:65)
        at de.itemis.mps.editor.celllayout.runtime.BorderPainter.paintBorders(BorderPainter.java:65)
        at de.itemis.mps.editor.celllayout.runtime.BorderPainter.paintBorders(BorderPainter.java:65)
        at de.itemis.mps.editor.celllayout.runtime.BorderPainter.paint(BorderPainter.java:35)
        at jetbrains.mps.nodeEditor.EditorComponent.paintComponent(EditorComponent.java:2232)
        at javax.swing.JComponent.paint(JComponent.java:1056)
        at jetbrains.mps.nodeEditor.EditorComponent.paint(EditorComponent.java:2562)
        at javax.swing.JComponent.paintChildren(JComponent.java:889)
        at javax.swing.JComponent.paint(JComponent.java:1065)
        at javax.swing.JViewport.paint(JViewport.java:728)
        at com.intellij.ui.components.JBViewport.paint(JBViewport.java:171)
        at javax.swing.JComponent.paintChildren(JComponent.java:889)
        at javax.swing.JComponent.paint(JComponent.java:1065)
        at javax.swing.JComponent.paintChildren(JComponent.java:889)
        at javax.swing.JComponent.paint(JComponent.java:1065)
        at javax.swing.JComponent.paintChildren(JComponent.java:889)
        at javax.swing.JComponent.paint(JComponent.java:1065)
        at javax.swing.JComponent.paintChildren(JComponent.java:889)
        at javax.swing.JComponent.paint(JComponent.java:1065)
        at javax.swing.JComponent.paintChildren(JComponent.java:889)
        at javax.swing.JComponent.paint(JComponent.java:1065)
        at javax.swing.JComponent.paintChildren(JComponent.java:889)
        at javax.swing.JComponent.paint(JComponent.java:1065)
        at javax.swing.JComponent.paintChildren(JComponent.java:889)
        at javax.swing.JComponent.paint(JComponent.java:1065)
        at javax.swing.JComponent.paintChildren(JComponent.java:889)
        at javax.swing.JComponent.paint(JComponent.java:1065)
        at javax.swing.JComponent.paintChildren(JComponent.java:889)
        at com.intellij.ui.tabs.impl.JBTabsImpl.paintChildren(JBTabsImpl.java:2316)
        at javax.swing.JComponent.paint(JComponent.java:1065)
        at com.intellij.ui.tabs.impl.JBTabsImpl.paint(JBTabsImpl.java:2311)
        at javax.swing.JComponent.paintChildren(JComponent.java:889)
        at javax.swing.JComponent.paint(JComponent.java:1065)
        at javax.swing.JComponent.paintChildren(JComponent.java:889)
        at javax.swing.JComponent.paint(JComponent.java:1065)
        at javax.swing.JComponent.paintChildren(JComponent.java:889)
        at javax.swing.JComponent.paint(JComponent.java:1065)
        at javax.swing.JComponent.paintChildren(JComponent.java:889)
        at javax.swing.JComponent.paint(JComponent.java:1065)
        at javax.swing.JComponent.paintChildren(JComponent.java:889)
        at javax.swing.JComponent.paint(JComponent.java:1065)
        at javax.swing.JComponent.paintChildren(JComponent.java:889)
        at javax.swing.JComponent.paint(JComponent.java:1065)
        at javax.swing.JLayeredPane.paint(JLayeredPane.java:586)
        at javax.swing.JComponent.paintChildren(JComponent.java:889)
        at com.intellij.openapi.wm.impl.ToolWindowsPane.paintChildren(ToolWindowsPane.java:190)
        at javax.swing.JComponent.paint(JComponent.java:1065)
        at javax.swing.JLayeredPane.paint(JLayeredPane.java:586)
        at javax.swing.JComponent.paintChildren(JComponent.java:889)
        at javax.swing.JComponent.paint(JComponent.java:1065)
        at javax.swing.JComponent.paintChildren(JComponent.java:889)
        at javax.swing.JComponent.paint(JComponent.java:1065)
        at javax.swing.JLayeredPane.paint(JLayeredPane.java:586)
        at javax.swing.JComponent.paintChildren(JComponent.java:889)
        at javax.swing.JComponent.paint(JComponent.java:1065)
        at javax.swing.JComponent.paintToOffscreen(JComponent.java:5210)
        at javax.swing.RepaintManager$PaintManager.paintDoubleBuffered(RepaintManager.java:1579)
        at javax.swing.RepaintManager$PaintManager.paint(RepaintManager.java:1502)
        at javax.swing.RepaintManager.paint(RepaintManager.java:1272)
        at javax.swing.JComponent._paintImmediately(JComponent.java:5158)
        at javax.swing.JComponent.paintImmediately(JComponent.java:4969)
        at javax.swing.RepaintManager$4.run(RepaintManager.java:831)
        at javax.swing.RepaintManager$4.run(RepaintManager.java:814)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
        at javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:814)
        at javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:789)
        at javax.swing.RepaintManager.prePaintDirtyRegions(RepaintManager.java:738)
        at javax.swing.RepaintManager.access$1200(RepaintManager.java:64)
        at javax.swing.RepaintManager$ProcessingRunnable.run(RepaintManager.java:1732)
        at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
        at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
        at java.awt.EventQueue.access$500(EventQueue.java:97)
        at java.awt.EventQueue$3.run(EventQueue.java:709)
        at java.awt.EventQueue$3.run(EventQueue.java:703)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
        at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
        at com.intellij.ide.IdeEventQueue.defaultDispatchEvent(IdeEventQueue.java:863)
        at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:699)
        at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:385)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
        at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
        at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

closed 2016-09-20 by slisson

vlsi, 2016-09-25:
Any reason why the issue has been closed?
As far as I understand, when that issue happens, the editor gets screwed.

Basically, if MPS launched with some previous project being open, then the exception might appear and mbeddr's editors will shown as reflective ones. The workaround is to rebuild languages, or to close all editor tabs and restart MPS.

I've filed the same exception against https://youtrack.jetbrains.com/issue/MPS-24656, however I'm not sure if that is mbeddr's or MPS' issue

slisson, 2016-09-26:
It was fixed in 437703da4721d36d6ccafff15809b521405653cd