mbeddr/mbeddr.core #1498: Strange ClassCastException with MPS 3.4

danielratiu, 2016-09-19:
java.lang.ClassCastException: java.awt.Color cannot be cast to java.lang.Boolean
        at de.itemis.mps.editor.celllayout.runtime.AbstractLayoutableAdapter.calcIsPushX(AbstractLayoutableAdapter.java:219)
        at de.itemis.mps.editor.celllayout.runtime.AbstractLayoutableAdapter.readSyncAll(AbstractLayoutableAdapter.java:171)
        at de.itemis.mps.editor.celllayout.runtime.LayoutableAdapter.readSyncAll(LayoutableAdapter.java:45)
        at de.itemis.mps.editor.celllayout.runtime.LayoutableCollectionAdapter.readSyncChildren(LayoutableCollectionAdapter.java:352)
        at de.itemis.mps.editor.celllayout.runtime.AbstractLayoutableAdapter.readSyncAll(AbstractLayoutableAdapter.java:166)
        at de.itemis.mps.editor.celllayout.runtime.LayoutableCollectionAdapter.readSyncAll(LayoutableCollectionAdapter.java:342)
        at de.itemis.mps.editor.celllayout.runtime.LayoutableCollectionAdapter.readSyncChildren(LayoutableCollectionAdapter.java:352)
        at de.itemis.mps.editor.celllayout.runtime.AbstractLayoutableAdapter.readSyncAll(AbstractLayoutableAdapter.java:166)
        at de.itemis.mps.editor.celllayout.runtime.LayoutableCollectionAdapter.readSyncAll(LayoutableCollectionAdapter.java:342)
        at de.itemis.mps.editor.celllayout.runtime.LayoutEngine$2.run(LayoutEngine.java:76)
        at de.itemis.mps.editor.celllayout.runtime.LayoutWatchdog.enter(LayoutWatchdog.java:21)
        at de.itemis.mps.editor.celllayout.runtime.LayoutEngine.layoutSubtreeUnsafe(LayoutEngine.java:71)
        at de.itemis.mps.editor.celllayout.runtime.LayoutEngine.layoutSubtree(LayoutEngine.java:66)
        at de.itemis.mps.editor.celllayout.runtime.LayoutEngine.layoutSubtree(LayoutEngine.java:47)
        at de.itemis.mps.editor.celllayout.runtime.LayoutEngine.layoutCell(LayoutEngine.java:43)
        at de.itemis.mps.editor.celllayout.runtime.LayoutInterceptor.doLayout(LayoutInterceptor.java:62)
        at jetbrains.mps.nodeEditor.cells.EditorCell_Collection.relayoutImpl(EditorCell_Collection.java:527)
        at jetbrains.mps.nodeEditor.cells.EditorCell_Basic.relayout(EditorCell_Basic.java:872)
        at jetbrains.mps.nodeEditor.EditorComponent.doRelayout(EditorComponent.java:1712)
        at jetbrains.mps.nodeEditor.EditorComponent.relayout(EditorComponent.java:1699)
        at de.itemis.mps.editor.celllayout.runtime.plugin.ProjectPlugin_CustomProjectPlugin$3.editorComponentCreate(ProjectPlugin_CustomProjectPlugin.java:43)
        at de.slisson.mps.hacks.editor.EditorComponentCreationListener.notifyCreate(EditorComponentCreationListener.java:61)
        at de.slisson.mps.hacks.editor.EditorComponentCreationListener.start(EditorComponentCreationListener.java:43)
        at de.itemis.mps.editor.celllayout.runtime.plugin.ProjectPlugin_CustomProjectPlugin.doInit(ProjectPlugin_CustomProjectPlugin.java:52)
        at jetbrains.mps.plugins.custom.BaseCustomProjectPlugin.init(BaseCustomProjectPlugin.java:46)
        at de.itemis.mps.editor.celllayout.runtime.plugin.Runtime_ProjectPlugin.initCustomParts(Runtime_ProjectPlugin.java:22)
        at jetbrains.mps.plugins.projectplugins.BaseProjectPlugin.initCustomParts1(BaseProjectPlugin.java:122)
        at jetbrains.mps.plugins.projectplugins.BaseProjectPlugin.init(BaseProjectPlugin.java:89)
        at jetbrains.mps.plugins.projectplugins.ProjectPluginManager.createPlugin(ProjectPluginManager.java:170)
        at jetbrains.mps.plugins.projectplugins.ProjectPluginManager.createPlugin(ProjectPluginManager.java:59)
        at jetbrains.mps.plugins.BasePluginManager.createPluginChecked(BasePluginManager.java:139)
        at jetbrains.mps.plugins.BasePluginManager.createPlugins(BasePluginManager.java:129)
        at jetbrains.mps.plugins.BasePluginManager.loadPlugins(BasePluginManager.java:76)
        at jetbrains.mps.plugins.projectplugins.ProjectPluginManager.loadPlugins(ProjectPluginManager.java:177)
        at jetbrains.mps.plugins.PluginLoaderRegistry.loadContributors(PluginLoaderRegistry.java:146)
        at jetbrains.mps.plugins.PluginLoaderRegistry.access$700(PluginLoaderRegistry.java:70)
        at jetbrains.mps.plugins.PluginLoaderRegistry$UpdatingTask.addContributors(PluginLoaderRegistry.java:334)
        at jetbrains.mps.plugins.PluginLoaderRegistry$UpdatingTask.doUpdate(PluginLoaderRegistry.java:324)
        at jetbrains.mps.plugins.PluginLoaderRegistry$UpdatingTask.lambda$run$0(PluginLoaderRegistry.java:310)
        at com.intellij.openapi.application.TransactionGuardImpl$6.run(TransactionGuardImpl.java:291)
        at com.intellij.openapi.application.impl.LaterInvocator$1.run(LaterInvocator.java:156)
        at com.intellij.openapi.application.impl.LaterInvocator$FlushQueue.runNextEvent(LaterInvocator.java:326)
        at com.intellij.openapi.application.impl.LaterInvocator$FlushQueue.run(LaterInvocator.java:310)
        at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
        at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
        at java.awt.EventQueue.access$500(EventQueue.java:97)
        at java.awt.EventQueue$3.run(EventQueue.java:709)
        at java.awt.EventQueue$3.run(EventQueue.java:703)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
        at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
        at com.intellij.ide.IdeEventQueue.defaultDispatchEvent(IdeEventQueue.java:863)
        at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:699)
        at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:385)
        at com.intellij.ide.IdeEventQueue.pumpEventsForHierarchy(IdeEventQueue.java:950)
        at com.intellij.openapi.progress.util.ProgressWindow.startBlocking(ProgressWindow.java:207)
        at com.intellij.openapi.progress.util.ProgressWindow.startBlocking(ProgressWindow.java:194)
        at com.intellij.openapi.application.impl.ApplicationImpl.runProcessWithProgressSynchronously(ApplicationImpl.java:569)
        at com.intellij.openapi.progress.impl.CoreProgressManager.runProcessWithProgressSynchronously(CoreProgressManager.java:396)
        at com.intellij.openapi.progress.impl.ProgressManagerImpl.runProcessWithProgressSynchronously(ProgressManagerImpl.java:76)
        at com.intellij.openapi.progress.impl.CoreProgressManager.run(CoreProgressManager.java:312)
        at jetbrains.mps.plugins.PluginLoaderRegistry.runTask(PluginLoaderRegistry.java:180)
        at jetbrains.mps.plugins.PluginLoaderRegistry.update(PluginLoaderRegistry.java:268)
        at com.intellij.openapi.application.TransactionGuardImpl$6.run(TransactionGuardImpl.java:291)
        at com.intellij.openapi.application.impl.LaterInvocator$FlushQueue.runNextEvent(LaterInvocator.java:326)
        at com.intellij.openapi.application.impl.LaterInvocator$FlushQueue.run(LaterInvocator.java:310)
        at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
        at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
        at java.awt.EventQueue.access$500(EventQueue.java:97)
        at java.awt.EventQueue$3.run(EventQueue.java:709)
        at java.awt.EventQueue$3.run(EventQueue.java:703)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
        at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
        at com.intellij.ide.IdeEventQueue.defaultDispatchEvent(IdeEventQueue.java:863)
        at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:699)
        at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:385)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
        at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
        at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

slisson, 2016-09-19:
This exception was already there in MPS 3.3. It's an MPS bug.

alexanderpann, 2023-02-19:
This is still an issue. I've commented on [MPS-33520](https://youtrack.jetbrains.com/issue/MPS-33520/) which deals with the same exception.