mbeddr/mbeddr.core #1500: diagram, MPS 3.4: InvalidDnDOperationException: Drag and drop in progress when doing several drag&drop with little delay in-between

vlsi, 2016-09-20:
```
java.awt.dnd.InvalidDnDOperationException: Drag and drop in progress
	at sun.awt.dnd.SunDragSourceContextPeer.setDragDropInProgress(SunDragSourceContextPeer.java:358)
	at java.awt.dnd.DragSource.startDrag(DragSource.java:301)
	at java.awt.dnd.DragSource.startDrag(DragSource.java:396)
	at java.awt.dnd.DragGestureEvent.startDrag(DragGestureEvent.java:292)
	at com.mxgraph.swing.handler.mxGraphHandler$3.dragGestureRecognized(mxGraphHandler.java:299)
	at java.awt.dnd.DragGestureRecognizer.fireDragGestureRecognized(DragGestureRecognizer.java:361)
	at sun.lwawt.macosx.CMouseDragGestureRecognizer.mouseDragged(CMouseDragGestureRecognizer.java:139)
	at java.awt.AWTEventMulticaster.mouseDragged(AWTEventMulticaster.java:320)
	at java.awt.AWTEventMulticaster.mouseDragged(AWTEventMulticaster.java:319)
	at java.awt.AWTEventMulticaster.mouseDragged(AWTEventMulticaster.java:319)
	at java.awt.AWTEventMulticaster.mouseDragged(AWTEventMulticaster.java:319)
	at java.awt.Component.processMouseMotionEvent(Component.java:6581)
	at javax.swing.JComponent.processMouseMotionEvent(JComponent.java:3342)
	at de.itemis.mps.editor.diagram.runtime.jgraph.MyGraphComponent$MyGraphControl.processMouseMotionEvent(MyGraphComponent.java:507)
	at java.awt.Component.processEvent(Component.java:6302)
	at java.awt.Container.processEvent(Container.java:2236)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2294)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4888)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4542)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4466)
	at java.awt.Container.dispatchEventImpl(Container.java:2280)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at com.intellij.ide.IdeEventQueue.defaultDispatchEvent(IdeEventQueue.java:863)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:695)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:385)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```

coolya, 2016-09-20:
Is this on linux?

vlsi, 2016-09-20:
That's macos 10.11.6 (15G1004)

slisson, 2016-09-20:
I have also seen that on OSX. It requires you to restart MPS.

alexanderpann, 2023-02-19:
I am closing because this issue should be fixed with https://github.com/JetBrains/MPS-extensions/commit/f355bb44a2b7ac1112b9beb402f33083c68fb9c9

closed 2023-02-19 by alexanderpann