mbeddr/mbeddr.core #1503: Edges can make boxes unselectable in graph editor with curved edges

eugenschindler, 2016-09-21:
[image]
Consider the situation above (an URL to the MPS model of this situation is housed in https://github.com/eugenschindler/mps-experiment under mps-diagram-editor): edge cb2 makes it so that Box C cannot be selected.
It is as if edge cb2 (if it were closed with a straight line between the first and last control point) becomes a planar object that obstructs C.
Could this please be fixed? I really don't know exactly where to start with a solution, but I can imagine that a fast quick-fix idea could be to move the edges to a lower layer (if the underlying library supports layers). Otherwise, maybe there is some setting or way to make the edges not behave as if they are planar objects, but really lines (maybe like a line without a fill?).

slisson, 2016-09-21:
Try the latest mbeddr version. I fixed this already.

eugenschindler, 2016-09-21:
Alright, thanks a lot! It seems I'm a bit behind. Time to upgrade :)

closed 2016-09-21 by slisson