mbeddr/mbeddr.core #1511: alias with array type and initialization will generate in two line.

huibo-shi, 2016-09-22:
When I tried to define a constant with uint8[4] type and initialization, then used it to fill another uint8[4] array, I found the generated code is not correct.

Here is the code source:
user_config module:
exported #alias uint8[4] SERVER_IP = {192, 168, 1, 101};

main module:
memcpy(server->ip, SERVER_IP, 4)

ip has type uint8[4]

Generated code:
user_config.h:
#define USER_CONFIG_SERVER_IP ({
  192,  168,  1,  101})  //generated code in two line.

main.c:
  memcpy(server->ip,USER_CONFIG_SERVER_IP,4); //as USER_CONFIG_SERVER_IP is not a pointer, there is a syntax error.

coolya, 2016-09-22:
This is not a problem of the generated code. You can't use compound literals as a source for memcpy. 

It expects a pointer.

huibo-shi, 2016-09-22:
Maybe I didn't describe very clear.  The code `exported #alias uint8[4] SERVER_IP = {192, 168, 1, 101}; `
will generate
```
#define USER_CONFIG_SERVER_IP ({
192, 168, 1, 101})
```
The generated code is in two lines. As mbeddr can accept USER_CONFIG_SERVER_IP as the parameter of memcpy without errors, and when we check USER_CONFIG_SERVER_IP's type in mbeddr, it has type uint8[4], so it's a pointer, but we generate compound literals. For me, this is also some kind of bugs. :(

coolya, 2016-09-22:
Indeed its a missing checking rule.