mbeddr/mbeddr.core #1515: Unit: Automatically convert units used in BinaryExpressions

slisson, 2016-09-23:
Example: I have two units _Minutes_ and _Seconds_ and a conversion rule between them. When I write an expression like `duration < 8 Minutes` where duration is in Seconds, the required conversion should be done automatically.

bkolb, 2016-09-23:
how do you check for overflow and precision loss if you do it automatically? I am not sure that automatic is the way to go. However there could be an quickfix that suggests the conversion

szabta89, 2016-09-23:
Actually, Bernd has a good point there. It did not come to my mind, but if the convert expression is not gonna be there, then the potential loss of precision will not be apparent to the user at all.

bkolb, 2016-09-23:
the original implementation did it automatically and we then changed it to the way we have it now for this reason

slisson, 2016-09-23:
What about applying only those conversions automatically, that use only operations that do not result in a precision loss (that contain only MulExpressions)

szabta89, 2016-09-23:
That would be an option. The question is how do we decide if the operation has the potential to lose precision. Simply checking whether the AST has a DivExpression? Would that be sufficient?

slisson, 2016-09-23:
A white list would be enough in our case.

slisson, 2016-09-23:
Or you add a flag to the conversion rule that decides whether the conversion can be applied automatically or not.

bkolb, 2016-09-23:
whitelist in which sense? for a unit?

slisson, 2016-09-23:
For the expressions that are allowed for automatically applied conversions. But we can also let the one who wrote the conversion decide that explicitly.

stephaneberle9, 2016-09-26:
In our smart meter project, we have quite a big catalog of very sophisticated unit conversions (see MetrologyUnitDeclarations for details). In many cases, we had to introduce explicit typecasts to avoid over/underflows. We also have provides alternative implementations of the same conversion rule depending on whether integer or double values are used to optimize division operations for speed. Over the time the conversion rules became so complex that we even started to implement unit tests to be sure that the rules do what we want and to avoid regressions (see MetrologyUnitDeclarationTests for details).

Whatever will boil out of this discussion, please keep in mind that the "automatic" conversion rule can only be a rough approximation to reality. They should only be offered as an alternative to manually tweaked conversion rules. The automatic rules can be an acceptable default in cases where users "don't care" about things like conversion precision and optimization. But for the very many cases where users do care about that stuff, there should be an obvious and flexible way to implement and choose customized conversion rules.

slisson, 2016-10-21:
Not required anymore

closed 2016-10-21 by slisson