mbeddr/mbeddr.core #1525: NPE when using actionsfilter

hnehls, 2016-09-28:
Hello,

any time I do a rebuild of my project, I get a NPE in the ActionsApplicationComponent.getInstance().

Interesting is that this NPE was not observed since the StandalonePluginDescriptor (without any modification, just the default) was added many weeks ago, but is now present in any rebuild action since we added the entire content of com.mbeddr.allInOne.zip (version nightly-127) to MPS\plugins folder.

Also strange seems for me: 
An 'Copy Node FQName' on the StandalonePluginDescriptor in Logical View returns 'com.siemens.ct.scanner.pluginSolution.plugin.null'. 
Why null?

Stacktrace:
```
java.lang.NullPointerException
	at com.mbeddr.mpsutil.actionsfilter.runtime.ActionsApplicationComponent.getInstance(ActionsApplicationComponent.java:26)
	at com.siemens.ct.scanner.projectView.pluginSolution.plugin.ActionModificationsApplicationPlugin_CustomApplicationPlugin.doInit(ActionModificationsApplicationPlugin_CustomApplicationPlugin.java:23)
	at jetbrains.mps.plugins.custom.BaseCustomApplicationPlugin.init(BaseCustomApplicationPlugin.java:26)
	at com.siemens.ct.scanner.projectView.pluginSolution.plugin.PluginSolution_ApplicationPlugin.initCustomParts(PluginSolution_ApplicationPlugin.java:32)
	at jetbrains.mps.plugins.applicationplugins.BaseApplicationPlugin.createCustomParts(BaseApplicationPlugin.java:131)
	at jetbrains.mps.plugins.applicationplugins.ApplicationPluginManager.createCustomParts(ApplicationPluginManager.java:113)
	at jetbrains.mps.plugins.applicationplugins.ApplicationPluginManager.afterPluginsCreated(ApplicationPluginManager.java:75)
	at jetbrains.mps.plugins.BasePluginManager.loadPlugins(BasePluginManager.java:76)
	at jetbrains.mps.plugins.PluginLoaderRegistry$2.run(PluginLoaderRegistry.java:130)
	at jetbrains.mps.smodel.WorkbenchModelAccess$6.compute(WorkbenchModelAccess.java:287)
	at jetbrains.mps.smodel.WorkbenchModelAccess$6.compute(WorkbenchModelAccess.java:284)
	at jetbrains.mps.util.ComputeRunnable.compute(ComputeRunnable.java:32)
	at jetbrains.mps.util.AbstractComputeRunnable.run(AbstractComputeRunnable.java:27)
	at jetbrains.mps.smodel.WriteActionDispatcher.run(WriteActionDispatcher.java:39)
	at jetbrains.mps.smodel.WriteActionDispatcher.compute(WriteActionDispatcher.java:49)
	at jetbrains.mps.smodel.WorkbenchModelAccess$7.compute(WorkbenchModelAccess.java:310)
	at jetbrains.mps.smodel.WorkbenchModelAccess$TryWriteActionComputable$1.compute(WorkbenchModelAccess.java:706)
	at com.intellij.openapi.application.impl.ApplicationImpl.runWriteAction(ApplicationImpl.java:1022)
	at jetbrains.mps.smodel.WorkbenchModelAccess$TryWriteActionComputable.compute(WorkbenchModelAccess.java:702)
	at jetbrains.mps.smodel.WorkbenchModelAccess.tryWrite(WorkbenchModelAccess.java:324)
	at jetbrains.mps.smodel.WorkbenchModelAccess.tryWrite(WorkbenchModelAccess.java:291)
	at jetbrains.mps.smodel.EDTExecutor$2.tryRun(EDTExecutor.java:78)
	at jetbrains.mps.smodel.EDTExecutor$Executor.worker(EDTExecutor.java:193)
	at jetbrains.mps.smodel.EDTExecutor$Executor.access$300(EDTExecutor.java:129)
	at jetbrains.mps.smodel.EDTExecutor$Executor$1.run(EDTExecutor.java:135)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
```

Any idea?

Best regards
Holger

slisson, 2016-10-06:
Is the plugin "com.mbeddr.mpsutil.actionsfilter.runtime" installed in Preferences -> Plugins?

hnehls, 2016-10-10:
Yes, it is.

alexanderpann, 2024-11-25:
So much has changed since the creation of the ticket. It is obsolete now.

closed 2024-11-25 by alexanderpann