mbeddr/mbeddr.core #1526: MarginComment overlaps editor content

hnehls, 2016-09-28:
When the editor content exceeds the 'virtual' page size, an added MarginComment overlaps the editor content:
[image]

slisson, 2016-09-29:
Not able able to reproduce it.

hnehls, 2016-09-29:
How can I support?
We are using MPS 3.3.5 and mbeddr nighlty_127. 
The 'overlapped editor content' is of type de.slisson.mps.tables filled with Expressions and the MarginComment is added to a table cell's content.
It is also visible when just displaying a very long name (INamedConcept.name).
Maybe it is depending on MPS version (was also present with 3.3.4) and is already solved with a later version but I could not check, yet.

bkolb, 2016-10-01:
Can you please provide an example so that we can debug it?

slisson, 2016-10-01:
I can reproduce it now. I didn't see the table.