mbeddr/mbeddr.core #1536: TreeNotation: exceptions on relayout

ftomassetti, 2016-10-07:
When opening models using the tree notation I frequently get exceptions similar to this one:

```
ERROR - or.cells.EditorCell_Collection - Some child cells of EditorCell_Collection cell with the layout: jetbrains.mps.nodeEditor.cellLayout.CellLayout_Indent@24d52152 still needs re-layout 
java.lang.Throwable
	at jetbrains.mps.nodeEditor.cells.EditorCell_Collection.relayoutImpl(EditorCell_Collection.java:507)
	at jetbrains.mps.nodeEditor.cells.EditorCell_Basic.relayout(EditorCell_Basic.java:847)
	at com.mbeddr.mpsutil.treenotation.runtime.TreeLayout$3.visit(TreeLayout.java:70)
	at com.mbeddr.mpsutil.treenotation.runtime.TreeLayout$3.visit(TreeLayout.java:68)
	at jetbrains.mps.internal.collections.runtime.IVisitor.invoke(IVisitor.java:12)
	at jetbrains.mps.internal.collections.runtime.IterableUtils.visitAll(IterableUtils.java:16)
	at jetbrains.mps.internal.collections.runtime.AbstractSequence.visitAll(AbstractSequence.java:80)
	at com.mbeddr.mpsutil.treenotation.runtime.TreeLayout.layout(TreeLayout.java:68)
	at com.mbeddr.mpsutil.treenotation.runtime.TreeLayout.doLayout(TreeLayout.java:37)
	at jetbrains.mps.nodeEditor.cells.EditorCell_Collection.relayoutImpl(EditorCell_Collection.java:502)
	at jetbrains.mps.nodeEditor.cells.EditorCell_Basic.relayout(EditorCell_Basic.java:847)
	at com.mbeddr.mpsutil.treenotation.runtime.TreeLayout$3.visit(TreeLayout.java:70)
	at com.mbeddr.mpsutil.treenotation.runtime.TreeLayout$3.visit(TreeLayout.java:68)
	at jetbrains.mps.internal.collections.runtime.IVisitor.invoke(IVisitor.java:12)
	at jetbrains.mps.internal.collections.runtime.IterableUtils.visitAll(IterableUtils.java:16)
	at jetbrains.mps.internal.collections.runtime.AbstractSequence.visitAll(AbstractSequence.java:80)
	at com.mbeddr.mpsutil.treenotation.runtime.TreeLayout.layout(TreeLayout.java:68)
	at com.mbeddr.mpsutil.treenotation.runtime.TreeLayout.doLayout(TreeLayout.java:37)
	at jetbrains.mps.nodeEditor.cells.EditorCell_Collection.relayoutImpl(EditorCell_Collection.java:502)
	at jetbrains.mps.nodeEditor.cells.EditorCell_Basic.relayout(EditorCell_Basic.java:847)
	at com.mbeddr.mpsutil.treenotation.runtime.TreeLayout$3.visit(TreeLayout.java:70)
	at com.mbeddr.mpsutil.treenotation.runtime.TreeLayout$3.visit(TreeLayout.java:68)
	at jetbrains.mps.internal.collections.runtime.IVisitor.invoke(IVisitor.java:12)
	at jetbrains.mps.internal.collections.runtime.IterableUtils.visitAll(IterableUtils.java:16)
	at jetbrains.mps.internal.collections.runtime.AbstractSequence.visitAll(AbstractSequence.java:80)
	at com.mbeddr.mpsutil.treenotation.runtime.TreeLayout.layout(TreeLayout.java:68)
	at com.mbeddr.mpsutil.treenotation.runtime.TreeLayout.doLayout(TreeLayout.java:37)
	at jetbrains.mps.nodeEditor.cells.EditorCell_Collection.relayoutImpl(EditorCell_Collection.java:502)
	at jetbrains.mps.nodeEditor.cells.EditorCell_Basic.relayout(EditorCell_Basic.java:847)
	at com.mbeddr.mpsutil.treenotation.runtime.TreeLayout$3.visit(TreeLayout.java:70)
	at com.mbeddr.mpsutil.treenotation.runtime.TreeLayout$3.visit(TreeLayout.java:68)
	at jetbrains.mps.internal.collections.runtime.IVisitor.invoke(IVisitor.java:12)
	at jetbrains.mps.internal.collections.runtime.IterableUtils.visitAll(IterableUtils.java:16)
	at jetbrains.mps.internal.collections.runtime.AbstractSequence.visitAll(AbstractSequence.java:80)
	at com.mbeddr.mpsutil.treenotation.runtime.TreeLayout.layout(TreeLayout.java:68)
	at com.mbeddr.mpsutil.treenotation.runtime.TreeLayout.doLayout(TreeLayout.java:37)
	at jetbrains.mps.nodeEditor.cells.EditorCell_Collection.relayoutImpl(EditorCell_Collection.java:502)
	at jetbrains.mps.nodeEditor.cells.EditorCell_Basic.relayout(EditorCell_Basic.java:847)
	at jetbrains.mps.nodeEditor.cellLayout.CellLayout_Vertical.doLayout(CellLayout_Vertical.java:69)
	at jetbrains.mps.nodeEditor.cells.EditorCell_Collection.relayoutImpl(EditorCell_Collection.java:502)
	at jetbrains.mps.nodeEditor.cells.EditorCell_Basic.relayout(EditorCell_Basic.java:847)
	at jetbrains.mps.nodeEditor.EditorComponent.doRelayout(EditorComponent.java:1788)
	at jetbrains.mps.nodeEditor.EditorComponent.relayout(EditorComponent.java:1775)
	at jetbrains.mps.nodeEditor.EditorContext$6.run(EditorContext.java:298)
	at jetbrains.mps.smodel.WorkbenchModelAccess.runReadAction(WorkbenchModelAccess.java:96)
	at jetbrains.mps.nodeEditor.EditorContext.setMemento(EditorContext.java:295)
	at jetbrains.mps.nodeEditor.EditorComponent.setRootCell(EditorComponent.java:1653)
	at jetbrains.mps.nodeEditor.updater.UpdaterImpl.doUpdate(UpdaterImpl.java:113)
	at jetbrains.mps.nodeEditor.updater.UpdaterImpl.update(UpdaterImpl.java:87)
	at jetbrains.mps.nodeEditor.EditorComponent.rebuildEditorContent(EditorComponent.java:1949)
	at jetbrains.mps.nodeEditor.NodeEditorComponent.rebuildEditorContent(NodeEditorComponent.java:110)
	at jetbrains.mps.nodeEditor.EditorComponent$27.run(EditorComponent.java:1164)
	at jetbrains.mps.smodel.WorkbenchModelAccess.runReadAction(WorkbenchModelAccess.java:96)
	at jetbrains.mps.smodel.ModelAccessBase.runReadAction(ModelAccessBase.java:53)
	at jetbrains.mps.nodeEditor.EditorComponent.editNode(EditorComponent.java:1127)
	at jetbrains.mps.ide.editor.BaseNodeEditor$1.performTask(BaseNodeEditor.java:110)
	at jetbrains.mps.ide.editor.BaseNodeEditor$PrioritizedTask.run(BaseNodeEditor.java:271)
	at jetbrains.mps.ide.editor.BaseNodeEditor.executeInEDT(BaseNodeEditor.java:125)
	at jetbrains.mps.ide.editor.BaseNodeEditor.editNode(BaseNodeEditor.java:103)
	at jetbrains.mps.ide.editor.NodeEditor.<init>(NodeEditor.java:36)
	at jetbrains.mps.ide.editor.MPSEditorOpener.createEditorFor(MPSEditorOpener.java:90)
	at jetbrains.mps.ide.editor.MPSEditorOpener.createEditorFor(MPSEditorOpener.java:72)
	at jetbrains.mps.ide.editor.MPSFileNodeEditor.recreateEditor(MPSFileNodeEditor.java:316)
	at jetbrains.mps.ide.editor.MPSFileNodeEditor$2.run(MPSFileNodeEditor.java:145)
	at jetbrains.mps.smodel.WorkbenchModelAccess.runReadAction(WorkbenchModelAccess.java:96)
[... 28 lines cut ...]
	at jetbrains.mps.smodel.WorkbenchModelAccess$TryWriteActionComputable.compute(WorkbenchModelAccess.java:702)
	at jetbrains.mps.smodel.WorkbenchModelAccess.tryWrite(WorkbenchModelAccess.java:324)
	at jetbrains.mps.smodel.WorkbenchModelAccess.tryWrite(WorkbenchModelAccess.java:291)
	at jetbrains.mps.smodel.EDTExecutor$2.tryRun(EDTExecutor.java:78)
	at jetbrains.mps.smodel.EDTExecutor$Executor.worker(EDTExecutor.java:193)
	at jetbrains.mps.smodel.EDTExecutor$Executor.access$300(EDTExecutor.java:129)
	at jetbrains.mps.smodel.EDTExecutor$Executor$1.run(EDTExecutor.java:135)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at com.intellij.ide.IdeEventQueue.defaultDispatchEvent(IdeEventQueue.java:866)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:654)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:381)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```

ftomassetti, 2016-12-14:
In certain cases the console is flooded by these exceptions. This is bad because it is make very hard to get important messages. Also, meaningful exceptions can easily get lost in the noise caused by this exception

slisson, 2016-12-14:
I don't know how to reproduce it. Do you use any special notations of cells inside the tree cells?

alexanderpann, 2024-07-14:
I haven't seen this exception with any real-world models, so it doesn't seem to be reproducible. @ftomassetti please reopen it if this is still an issue and you have steps to reproduce it.

closed 2024-07-14 by alexanderpann