mbeddr/mbeddr.core #1537: IllegalModelAccessError in contextactions

danielratiu, 2016-10-11:
<null> : You can read model only inside read actions
jetbrains.mps.smodel.IllegalModelAccessError: You can read model only inside read actions
	at jetbrains.mps.smodel.WorkbenchModelAccess.checkReadAccess(WorkbenchModelAccess.java:473)
	at jetbrains.mps.smodel.ModelAccessBase.checkReadAccess(ModelAccessBase.java:38)
	at jetbrains.mps.smodel.MPSModuleRepository.getModules(MPSModuleRepository.java:279)
	at com.mbeddr.mpsutil.contextactions.runtime.DescriptorCollector.start(DescriptorCollector.java:41)
	at com.mbeddr.mpsutil.contextactions.runtime.ContextActionManager.init(ContextActionManager.java:35)
	at com.mbeddr.mpsutil.contextactions.runtime.plugin.ApplicationPlugin_CustomApplicationPlugin.doInit(ApplicationPlugin_CustomApplicationPlugin.java:12)
	at jetbrains.mps.plugins.custom.BaseCustomApplicationPlugin.init(BaseCustomApplicationPlugin.java:26)
	at com.mbeddr.mpsutil.contextactions.runtime.plugin.Runtime_ApplicationPlugin.initCustomParts(Runtime_ApplicationPlugin.java:24)
	at jetbrains.mps.plugins.applicationplugins.BaseApplicationPlugin.createCustomParts(BaseApplicationPlugin.java:131)
	at jetbrains.mps.plugins.applicationplugins.ApplicationPluginManager.createCustomParts(ApplicationPluginManager.java:112)
	at jetbrains.mps.plugins.applicationplugins.ApplicationPluginManager.afterPluginsCreated(ApplicationPluginManager.java:74)
	at jetbrains.mps.plugins.BasePluginManager.loadPlugins(BasePluginManager.java:88)
	at jetbrains.mps.plugins.PluginLoaderRegistry.loadContributors(PluginLoaderRegistry.java:146)
	at jetbrains.mps.plugins.PluginLoaderRegistry.access$700(PluginLoaderRegistry.java:70)
	at jetbrains.mps.plugins.PluginLoaderRegistry$UpdatingTask.addContributors(PluginLoaderRegistry.java:334)
	at jetbrains.mps.plugins.PluginLoaderRegistry$UpdatingTask.doUpdate(PluginLoaderRegistry.java:324)
	at jetbrains.mps.plugins.PluginLoaderRegistry$UpdatingTask.lambda$run$0(PluginLoaderRegistry.java:310)
	at com.intellij.openapi.application.TransactionGuardImpl$6.run(TransactionGuardImpl.java:291)
	at com.intellij.openapi.application.impl.LaterInvocator$1.run(LaterInvocator.java:156)
	at com.intellij.openapi.application.impl.LaterInvocator$FlushQueue.runNextEvent(LaterInvocator.java:326)
	at com.intellij.openapi.application.impl.LaterInvocator$FlushQueue.run(LaterInvocator.java:310)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at com.intellij.ide.IdeEventQueue.defaultDispatchEvent(IdeEventQueue.java:863)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:699)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:385)
	at com.intellij.ide.IdeEventQueue.pumpEventsForHierarchy(IdeEventQueue.java:950)
	at com.intellij.openapi.progress.util.ProgressWindow.startBlocking(ProgressWindow.java:207)
	at com.intellij.openapi.progress.util.ProgressWindow.startBlocking(ProgressWindow.java:194)
	at com.intellij.openapi.application.impl.ApplicationImpl.runProcessWithProgressSynchronously(ApplicationImpl.java:569)
	at com.intellij.openapi.progress.impl.CoreProgressManager.runProcessWithProgressSynchronously(CoreProgressManager.java:396)
	at com.intellij.openapi.progress.impl.ProgressManagerImpl.runProcessWithProgressSynchronously(ProgressManagerImpl.java:76)
	at com.intellij.openapi.progress.impl.CoreProgressManager.run(CoreProgressManager.java:312)
	at jetbrains.mps.plugins.PluginLoaderRegistry.runTask(PluginLoaderRegistry.java:180)
	at jetbrains.mps.plugins.PluginLoaderRegistry.update(PluginLoaderRegistry.java:268)
	at com.intellij.openapi.application.TransactionGuardImpl$6.run(TransactionGuardImpl.java:291)
	at com.intellij.openapi.application.impl.LaterInvocator$FlushQueue.runNextEvent(LaterInvocator.java:326)
	at com.intellij.openapi.application.impl.LaterInvocator$FlushQueue.run(LaterInvocator.java:310)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at com.intellij.ide.IdeEventQueue.defaultDispatchEvent(IdeEventQueue.java:863)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:699)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:385)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

DomenikP, 2016-10-11:
Do you get this exception with MPS 3.4?

danielratiu, 2016-10-11:
yes - milestone 15.3

commit 42f84af2 (2016-10-25): ContextActions: IllegalModelAccessError in contextactions #1537

Fixes #1537
closed 2016-10-25 by slisson