mbeddr/mbeddr.core #1569: Issue with splittable and tokenizer function

ftomassetti, 2016-11-04:
I am using a splittable cell with a tokenizer function containing a print statement.
I was expecting that while typing the tokenizer function would be invoked. Apparently it is not.
Am I missing how splittable work? When the tokenizer should be invoked? Is it a bug?

slisson, 2016-11-04:
You need a property constraint. The tokenizer is invoked, when the user enters a value, that is not valid for the property.

ftomassetti, 2016-11-04:
ah, I see! Thank you. Would it make sense to show a warning when the property constraint is not present?

slisson, 2016-11-04:
Yes, I will add that.

ftomassetti, 2016-11-07:
Also, now the tokenizer function is invoked and calculate different parts for my original string. However it is not clear what should happen: the cell now shows an error and contain the whole string anyway.

I have `foo` I type `$` and the tokenizer produces `[foo, $]` however the cell keeps showing `foo$`

slisson, 2016-11-07:
If nothing happens, it means that the parser cannot find a parse tree.

closed 2017-01-13 by slisson