mbeddr/mbeddr.core #1577: Too constrained handling of EnumTypes

danielratiu, 2016-11-11:
I have the following code:
[image]

and mbeddr reports a type error. I would like to fix this by enhancing superTypes_of_enumType with:
[image]

(according to the C standard, enums can have in C up to 1023 literals)

bkolb, 2016-11-11:
No, I disagree. I think there should be an explicit cast

danielratiu, 2016-11-12:
My thinking is that:
1) this is a clean C programming pattern
2) we do not provide a better solution in mbeddr for the example I gave 
3) we have the information in mbeddr about the ranges of enums (and can easily compute their type)
4) using casts everywhere would bloat the code unnecessarily and thereby harden the migration

bkolb, 2016-11-13:
I fundamentally disagree.

reg 1): there are many _clean_ C patterns that we do not support for good reasons. Treating Enums as regular ints has problems in various places and looses semantics. 

reg 2): you can use constants. Or casts

reg 3): I don't know how that would help. Do you want to allow an Enum as lower or upper bound in an loop e.g.? I think that makes no sense what so ever. And: With your proposal an enum with and without init expression would behave differently. 

reg 4) Harden what? During migration you can check if your enum has semantic information or not. If it does, most likely it should not be treated as an int. 

If we do what you propose, you can use enums (without literals with init expressions) in all places where ints are allowed. E.g. in Arithmetic operations. I think that does not make sense. 
*IF* you want to do something, then add a `.value` DotTarget for enumtypes. That can be used to get the underlying value and would during generation just be abandoned. It would be not that much different from a cast though.

danielratiu, 2016-11-13:
> .value DotTarget for enumtypes
this is a good idea

coolya, 2016-11-14:
My two cents for this:

1)
I don't think this is clean practice at all, especially the case that you are showing here is dangerous. If an enum member is added after the literal that you use to declare the array. That's why many C code bases feature a last member in the `enum` that has no semantical meaning except that "max value". Having clean enum semantics is something we should aim for.

2)
as @bkolb mentioned there is some other ways to work with that.

3)
This would imply we handle enums as integer constants with no additional semantics. I don't think we want to do this. One of the problems in C with enums is their very weak semantics.

4)
I think this can be mitigated by having a semi automatic migration. 

I don't see how `.value` is better than a cast, but if users can deal with it easier ... why not. But there is a second thing we need from my point of view. And that is a way to get the maximum integer value of a enum type. As you would need to make your code example work the correct way.