mbeddr/mbeddr.core #1582: reported error in editingGuide.Exercise - how to exclude the editingGuide-solution from model checker?

hnehls, 2016-11-18:
I created a short tutorial using the editingGuide feature. The user shall complete a statement by using the completion menu. So far so good. 
Now there is a checking rule to check for the mandatory instance (the one that is created by the user of the tutorial later on). This check always reports an error when running the model checker (e.g. git commit). And when continue with this tutorial, the number of errors may increase.

Is it possible to exclude the editingGuide from the model checker or suppress the error checks within an editingGuide.Exercise?

coolya, 2016-11-18:
There is the `ISuppressErrors` interface in `jetbrains.mps.lang.core` which you can implement in a NodeAttribute and then attach it to the nodes in the editing guide model. Have a look at for instance `ReferenceMacro` in the generator language which basically does the same. 

This will prevent the ErrorReporter from reporting it.

alexanderpann, 2022-04-27:
Closing because question was answered.

closed 2022-04-27 by alexanderpann