mbeddr/mbeddr.core #1592: log expression doesn't add '%lld' formater for uint64

huibo-shi, 2016-12-05:
Hi, 

Currently in mbeddr the log expression doesn't behave correctly for uint64.

`uint64 size;
log [size: @expr(size)]` will generate
`uint64_t size;
printf("size: %d \n", size);`

However it should be `printf("size: %lld \n", size);` Otherwise the value print is always 0.

fbacha, 2016-12-22:
see http://www.cplusplus.com/reference/cstdio/printf/?kw=printf   for complete reference of format specifiers

coolya, 2017-01-10:
@fbacha @huibo-shi Do you need the bugfix in the MPS 3.3 branch or is ok if i fix it on master?

fbacha, 2017-01-10:
@coolya We have already migrated to MPS 3.4.2, so it's OK for the fix on the master only.
Thanks!

coolya, 2017-01-10:
I have added a lot more missing format strings based on the website and also added some tests.:

http://localhost:8080/select/com.mbeddr.core/r:0c020a2a-0fae-4359-90ba-63fdf45caa94/5760222092161279811/

commit bbf709dc (2017-01-10): core: Implement getPrintFormatString for uint 64 correctly

fixes #1592
closed 2017-01-10 by coolya